lgsvl/simulator #1369: Dallara IL-15 model preview not loading

sisaha9, 2021-04-15:
[image]

Every other model was loading fine for me but Dallara IL-15 doesn't. Is this an issue with the model or am I doing something wrong?

lemketron, 2021-04-15:
There may be an issue with that model. I have reproduced the problem and will file a bug for this. Thanks!

In addition, I was unable to create a new sensor configuration from the [sample IL-15 ROS2 sensor JSON file](https://content.lgsvlsimulator.com/vehicles/dallara-ros2-sensors.json) that I created for the 2020.06 simulator. 

We will investigate and ideally add (or share) the sample Autoware.Auto sensor config for the 2021.1 simulator.

EricBoiseLGSVL, 2021-04-16:
Yes, looks like our system missed something building it.  We will need to rebuild.

revati-naik, 2021-07-08:
Looks like this issue has been resolved.

closed 2021-07-08 by revati-naik