lgsvl/simulator #1370: VSE with Apollo 6.0

1Jyfc, 2021-04-15:
Hello!
I'm recently working on Apollo 6.0 test and I tried to use VSE to create some testcases.
I've watched [this video](https://www.youtube.com/watch?v=9Fneu0PTEC4) on Youtube and then follow it to create a VSE json file. I'd like to try if simulation with VSE can connect to Apollo 6.0, but meet some trouble. Here is my steps:

For Apollo 6.0, I use the version `05854798ba065f597a50d4390d3cc770181c2780` which was updated on Mar 17, 2021. I followed the steps Apollo provided:
`./docker/scripts/dev_start.sh`
`./docker/scripts/dev_into.sh`
`./scripts/bootstrap_lgsvl.sh`
`./scripts/bridge.sh`
, and here is my Dreamview:
[image]

For SVL, here is my VSE config:
[image]
I use source code to build SVL simulator on Ubuntu 18.04, and use VSE to create the testcase:
[image]
and I run Apollo and SVL on the same computer, so the connection is localhost:9090. 

After that I tried a new Apollo 6.0 simulation with random traffic, which runs OK, so I think my Apollo is in good status. Is there anything I can do to run the VSE simulation? Hope for your help, thanks.

lemketron, 2021-04-15:
I believe that Perception and Traffic Light modules are not working in Apollo 6.0 which is fine as you are using "modular testing" sensor configuration which supplies the output of those components in Apollo so they are not needed. However, I noticed in your screen shot that you have Traffic Light module enabled. You should avoid enabling that module.

In addition, the text appearing in the development console shows that there was an error launching the VSE runner, and also indicates that you're running a Development build... I don't believe the VSE runner will work when you are running a development build of the simulator as the runner lives in a container that is delivered with the release build of the simulator.

There may be more information available in the log (click "Open Log File") but have you tried running the VSE scenario with a release (not source) build of the simulator?

1Jyfc, 2021-04-16:
Thanks for your reply! I build the release version of SVL and create a new VSE json file with it. However I got the same result with error "Error on latest run: Failed to launch TestCase runtime", and this time there is no log because of the release version. I think there's something wrong when I build simulator.

And finally I try the release version on github, and this time there's another wrong: 
Error on latest run: Test case exit code: 1 error data: 2021-04-16 11:26:35,469 [INFO][__main__] Run VSE scenario scenario.json 2021-04-16 11:26:35,507 [INFO][run_vse] Loading BorregasAve map. 2021-04-16 11:26:37,431 [INFO][run_vse] Loaded 1 ego agents 2021-04-16 11:26:37,431 [INFO][run_vse] Loaded 3 NPC agents 2021-04-16 11:26:37,431 [INFO][run_vse] Loaded 0 pedestrian agents 2021-04-16 11:26:39,960 [ERROR][dreamview] HD Map BorregasAve was not set. Verify the files exist in /apollo/modules/map/data/BorregasAve and restart Dreamview -- Aborting...

The folder /apollo/modules/map/data is here:
[image]

Is it means I should manually download a map and put it in this folder? Or is there anything wrong? I think if I need to download the map, in random traffic model there should be the same problem, but it runs OK...

1Jyfc, 2021-04-20:
Hello!
These days I try to change the folder name of `/apollo/modules/calibration/data/Lincoln2017MKZ_LGSVL` into `/apollo/modules/calibration/data/Lincoln2017MKZ` and it finally connects correctly. But I found some new problems. Hope for you reply.

Your comment several days ago said that if I use "module testing" sensor configuration, I should avoid enabling Traffic Light module (as well as Perception, I think). But after using the simple method of renaming the folder, each time I run the simulator, all the modules used will be autonomously enabled, including Traffic Light and Perception:
[image]

And the running status of apollo is unnormal, different from Random Traffic module. It seems that routing module can't run. Here are some files that may help, including logs of some modules and my VSE json file.
[useful_files.zip](https://github.com/lgsvl/simulator/files/6342308/useful_files.zip)

And this is the connection status, seems correct:
[image]

I think my vehicle config is wrong, is there any suggestion? Thanks.

lemketron, 2021-04-23:
> [ERROR][dreamview] HD Map BorregasAve was not set. Verify the files exist in /apollo/modules/map/data/BorregasAve and restart Dreamview

This is a good catch. The Borregas Ave map that is in upstream Apollo is in a directory called "borregas_ave" which is processed by DreamView to show up as "Borregas Ave" in the map menu in the web interface.

However, in order to make it easier to translate simulator map names to apollo map names for automated scenario testing, it is now needed to use a CamelCase version of the map name like "BorregasAve" as noted in the error message you shared. You can either copy or rename the `/apollo/modules/map/data/borregas_ave` directory to `BorregasAve` and try again.

As for the vehicle configurations, I don't believe you should change those directories. You should select the LGSVL variants, though the VSE scenario runner should actually select these for you.

Please note that the VSE runner is currently supported only on Linux. Please refer to the following for more information:
https://www.svlsimulator.com/docs/running-simulations/running-simulator/#vsesimulation

> each time I run the simulator, all the modules used will be autonomously enabled, including Traffic Light and Perception

I believe this is because you're using the wrong sensor configuration for the vehicle. Please edit the Simulation, go to the Test Case view, and make sure that under vehicle you have selected "Apollo 6.0 (modular testing)". This should prevent the perception and traffic signal modules from being enabled (and if it does not then that is a bug in the VSE Runner).

Also note (as documented): "The TestCaseRunner for VSE won't turn off apollo modules when the scenario finishes. You need to disable the modules once you finish the testing" -- this can cause problems when re-running a scenario as the Apollo Planning module may need to be disabled and re-enabled before it can plan a new route after the ego vehicle appears at (or is reset to) a new (start) location.

1Jyfc, 2021-04-26:
Thanks for reply.
For map configuration, the way of changing directories is useful. But for vehicle configurations:

> As for the vehicle configurations, I don't believe you should change those directories. You should select the LGSVL variants, though the VSE scenario runner should actually select these for you.

I don't understand what you mean. Do you mean that I should choose "Lincoln2017MKZ LGSVL" in Dreamview like this:
[image]

But as the picture shows, VSE will select "Lincoln2017MKZ", not the right one.

And for the Test Case configuration:
[image]

I think that's right, so maybe it's really a bug...

GabrielZZZ, 2021-04-28:
@1Jyfc I met the same problem as you

GabrielZZZ, 2021-04-28:
I encountered the same problem as @1Jyfc had, and I copied the configuration of "Lincoln2017MKZ_LGSVL" to "Lincoln2017MKZ" to make the simulator work. So far I did not see any problems related to modules possibly because I have set the destination point in the Visual Editor already. 

> Also note (as documented): "The TestCaseRunner for VSE won't turn off apollo modules when the scenario finishes. You need to disable the modules once you finish the testing" -- this can cause [...]

But the perception and traffic light module did turn on automatically even all the sensor configuration has been set to modular testing (every time I run the simulation these modules will turn off and on automatically so I do not need to manually enable / disable them).

ijklmnopqr, 2022-03-21:
Can anyone please share the working project of VSE with either Apollo 5.0 or Apollo 6.0.