lgsvl/simulator #1371: Invalid angular steering via ROS2 bridge in SVL and LGSVL Control Sensor

pijaro, 2021-04-15:
**System info** 
- Ubuntu 20.04
- SVL 2021.01

**Steps to reproduce issue**
- run simulation with `ROS2` bridge and vehicle `AWFLexus2016RXHybrid`
- send vehicle control command with `target_wheel_angle != 0`:
```shell
ros2 topic pub /lgsvl/vehicle_control_cmd lgsvl_msgs/msg/VehicleControlData  \
"{acceleration_pct: 0.5, target_wheel_angle: 0.6, target_wheel_angular_rate: 1.0, target_gear: 1}" -r 10
```

**Expected behavior**
Simulated car wheel angle matches `target_wheel_angle` in a published message.

**Actual behavior**
Car wheel angle doesn't match `target_wheel_angle`. Angle is noticeable lower than expected.

pijaro, 2021-04-15:
This bug is caused by redundant clamps of `steeringInput` and mismatched units. Redundancy can be found [here (LGSVL control sensor)](https://github.com/lgsvl/LGSVLControlSensor/blob/master/LGSVLControlSensor.cs#L130) and [here (ROS2 bridge)](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros2/Ros2Conversions.cs#L375). 

We can also see that ROS2 bridge returns `rad`s while `LGSVLControlSensor` compares `wheelAngle` with `degrees`.

pijaro, 2021-05-19:
@hadiTab any progress on this issue?

hadiTab, 2021-06-02:
@pijaro sorry I missed these. The steering issue was addressed shortly after the release. We elected not to change the code in the core simulator and instead changed code in the vehicle control sensor to fix it. The lgsvl vehicle control sensor in wise (https://wise.svlsimulator.com/plugins/profile/a5696615-d3e2-45e9-afa3-9b5ae7359e02) has been rebuilt and updated so the sensor should be redownloaded on next use and should be working properly now. Are you still having issues with this?

MertClk, 2021-07-12:
With this fix, is it expected to match `target_wheel_angle` in the control message and `steer_pct` in the state report message? I can not be sure that if the sent command is applied to the simulated vehicle since the state report message shows different steering values.

lemketron, 2021-08-24:
On [Autoware's gitlab issue 977](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/977#note_604050649), Piotr Jaroszek reported:

> I can confirm that angular steering is no longer an issue. Wheel angles looks ok now!

closed 2021-08-24 by lemketron