lgsvl/simulator #1373: Inconsistent gear enums for lgsvl_msgs and vehicle_state_sensor

pijaro, 2021-04-15:
**System info** 
- Ubuntu 20.04
- SVL 2021.1

**Steps to reproduce issue**
- run simulation with `ROS2` bridge and vehicle `AWFLexus2016RXHybrid`
- send vehicle state command `current_gear = 1` (_GEAR_DRIVE_) as defined [here (VehicleStateData.msg)](https://github.com/lgsvl/lgsvl_msgs/blob/master/msg/VehicleStateData.msg#L27)
```shell
ros2 topic pub /lgsvl/vehicle_state_cmd lgsvl_msgs/msg/VehicleStateData "current_gear: 1"
```

**Expected behavior**
Gearbox sets to "drive".

**Actual behavior**
Gearbox sets to "reverse".

pijaro, 2021-04-15:
This is because `VehicleStateSensor` actually `ShiftReverse` for `Gear == 1` as you can [see here](https://github.com/lgsvl/VehicleStateSensor/blob/fd155fcb5bf9c29c70703cb9d90356c235e17d02/VehicleStateSensor.cs#L73).

Additionally `ShiftFirstGear` is for `Gear == 0` which is also a mismatch.

pijaro, 2021-05-19:
@hadiTab can you look into this issue please?

hadiTab, 2021-06-02:
@pijaro sorry this has been sitting in our backlog but we haven't been able to get to it yet. We'll get someone working on it soon. Thanks.

mslavescu, 2021-06-25:
Hi @hadiTab, do you have any update on this issue?

lemketron, 2021-08-24:
On [Autoware's gitlab issue 977](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/977#note_604050649), Piotr Jaroszek reported:

> I can confirm that angular steering is no longer an issue. Wheel angles looks ok now!

EricBoiseLGSVL, 2021-09-01:
This is fixed in the next update right @hadiTab ?