lgsvl/simulator #1374: Can't open the simulator on unity

rythm07, 2021-04-15:
Hi,

Somehow when I git clone this repository and try to open it with Unity Editor 2019.4.18 on Linux, it starts loading and then closes without even opening the Unity editor. 
I've already tried to uninstall and install everything again but the "error" persists. Did anyone had this issue before? Are anyone able to help me with that? 
I want to build a map importing a point cloud and test it on the simulator but without it opening, I'm really stuck. Plus, if I start a new projection Unity Hub, the editor works just fine, but in that way, I cannot have access to the simulator capabilities.

Here is the link that I recorded if anyone didn't understand what's happening.
https://www.loom.com/share/430baf409eb841f790ade1398e991594

Thanks for the help!

EricBoiseLGSVL, 2021-04-16:
Please check that you are using a GPU to run simulator.  If you can grab the player log it would help to debug

rythm07, 2021-04-16:
Thanks for the response!

How can I check that if I'm using the GPU to run the simulator?

Here are the logs:
[image]

rythm07, 2021-04-16:
> Thanks for the response!
> 
> How can I check if I'm using the GPU to run the simulator?
> 
> Here are the logs:
> [image]
> [image]
> [image]
> [image]

EricBoiseLGSVL, 2021-04-16:
Are you using Xserver?  If so, is it using a GPU?

rythm07, 2021-04-16:
I think I'm not using Xserver.
The logs file is here. Sorry for the screenshots :)
[Editor.log](https://github.com/lgsvl/simulator/files/6328172/Editor.log)

EricBoiseLGSVL, 2021-04-17:
```
Gdk-Message: 23:42:32.018: Unity: Fatal IO error 9 (Bad file descriptor) on X server :1.
```
seems like it is trying to use x server and failing.  This also looks could be errors trying to download unity packages.  Do you have antivirus or company firewall blocking?

What GPU and kernel are you using?  I recommend installing latest drivers if you can

rythm07, 2021-04-17:
I tried to change to the X.Org X Server but It does the same thing.
[image]

No, I'm on Linux so I don't have an antivirus neither a company firewall blocking I think.
I'm using an RTX2060 GPU and I have the latest Nvidia drivers. The kernel version is Linux 5.8.0-50-generic x86_64.
[image]

EricBoiseLGSVL, 2021-04-19:
What version of Ubuntu are you on?

rythm07, 2021-04-21:
Yes I have mesa drivers installed. 

I'm using ubuntu 20.04 LTS.

EricBoiseLGSVL, 2021-04-22:
you must uninstall mesa drivers.  We have a few closed issues regarding this.

lemketron, 2021-04-22:
Even if you have installed libvulkan, if you also have mesa-vulkan-drivers installed then that can cause problems.

Please take a look at the Player.log file from `~/.config/unity3d/LGElectronics/SVLSimulator/Player.log` as there are often clues in that log file.

In particular, if you see "mesa" or "Intel" mentioned in Vulkan vendor or renderer lines rather than NVIDIA, then you might have "mesa-vulkan-drivers" installed. If so, then you can remove these with `sudo apt remove mesa-vulkan-drivers`.

rythm07, 2021-04-22:
Uninstalled mesa-vulkan-drivers and it worked! 

Thanks a lot for your help!

closed 2021-04-22 by rythm07