lgsvl/simulator #1375: Is it possible to load existing asset bundles into Unity for editing?

sisaha9, 2021-04-16:


EricBoiseLGSVL, 2021-04-16:
We provide assets pre-built and these cannot be edited.  We also provide open source repos of some assets and these are the ones you can edit.  What assets are you trying to edit?

sisaha9, 2021-04-16:
Ideally the Dallara one after the fix. I had made my track based off real world measurements and most of the cars except for the F1Tenth one take up the entire road

EricBoiseLGSVL, 2021-04-16:
Wait.  You want the dallara as a small vehicle. @lemketron didn't we only release the F1Tenth vehicle for this scale?

sisaha9, 2021-04-16:
The F1Tenth is a bit too small too haha. Was looking to scale the wheelbase by 3 and remaining factors by 2. It's fine either way. Was just curious if it was possible. If not I can clone the Jaguar repo, scale it down and work with it

EricBoiseLGSVL, 2021-04-16:
We can make the adjustments in the f1Tenth.  Do you have any reference docs or images to make the adjustments?

sisaha9, 2021-04-17:
I wanted to replicate the dimensions on this kart: http://topkartusa.net/electric/

lemketron, 2021-04-18:
We know folks working with the EV Kart; I will check and see if they have a vehicle model they can share in the new 2021 public library...

closed 2021-07-16 by EricBoiseLGSVL