lgsvl/simulator #1376: Problems with generating map in lgsimulator

lovebigwei, 2021-04-16:
HI, I am following the guide to import hd map from apollo5.0 to lgsimulator and generate it's correspondent map used in lg. I have build the map using unity. Bet when I upload the map to lg, it says the folloing error.
Unexpected error occurred. Please try again later or contact administrator
[image]

EricBoiseLGSVL, 2021-04-16:
can you post the asset bundle here so I can look?

lovebigwei, 2021-04-16:
> can you post the asset bundle here so I can look? or open in zip program and screen shot so I can see the contents?

[image]
[image]
here it is

lovebigwei, 2021-04-16:
or may be you are refering to this ?
[image]

EricBoiseLGSVL, 2021-04-16:
Open environment_newmap so we can see if the required data is correctly built.

lovebigwei, 2021-04-19:
> Open environment_newmap so we can see if the required data is correctly built. Also, details on the size of the file helps.

When I try to extract files in environment_newmap, it says error ,the size of the file is 445.0 kB.
[image]
[image]

EricBoiseLGSVL, 2021-04-19:
Can you post the file here?  If not, use 7zip to look inside without extracting

lemketron, 2021-04-23:
This sounds like it may be related to https://github.com/lgsvl/simulator/issues/1364 (and tracked as AUTO-6845).

closed 2021-07-16 by EricBoiseLGSVL