lgsvl/simulator #1379: simulator.exe cannot be opened

tk-s05, 2021-04-19:
Hi,
I can't launch the simulator.
I get an error when I try to start the simulator.　
・I uninstalled the simulator but couldn't start it
・I can launch it with other user accounts.
[image]
 Hope for your help, thanks.

EricBoiseLGSVL, 2021-04-20:
Please post OS, version of simulator, gpu and then post player.log so we can see the error.

tk-s05, 2021-04-21:
OS:Win10, GPU:Quadro P6000, version:2021.1 release

I downloaded from the SVL Simulator homepage and unzipped the file.
I used to be able to start it, but suddenly I can't start it.
Is there any other information needed to resolve this issue?

＜windows log＞
------------------------------------------------------------------------------------------------------------------------
Log name: Application
Source: Application Error
Date: 2021/04/21 13:43:49
Event ID: 1000
Task category: (100)
Level: Error
Keywords: classic
User: N / A
Computer:　（Keep it private）
Description:
Failed application name: simulator.exe, version: 2019.4.18.4260, timestamp: 0x5ffdc692
Failed module name: UnityPlayer.dll, version: 2019.4.18.4260, timestamp: 0x5ffdc82f
Exception code: 0xc0000005
Fault offset: 0x00000000009234c4
Failing process ID: 0xed90
Failed application start time: 0x01d73668bde2d55c
Failed application path: C: \ Users \ lgsvl \ Desktop \ svlsimulator-windows64-2021.1 \ simulator.exe
Failed module path: C: \ Users \ lgsvl \ Desktop \ svlsimulator-windows64-2021.1 \ UnityPlayer.dll
Report ID: caf10a98-c798-4176-8530-db7f067e7057
The full name of the failing package:
Application ID associated with the failing package:

tk-s05, 2021-04-21:
player.log
-----------------------------------------------------------------------------------------------------------
Mono path[0] = 'C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/Managed'
Mono config path = 'C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/MonoBleedingEdge/etc'
Loading player data from C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/data.unity3d
Initialize engine version: 2019.4.18f1 (3310a4d4f880)
[Subsystems] Discovering subsystems at path C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: NVIDIA Quadro P6000 (ID=0x1b30)
    Vendor:   
    VRAM:     24431 MB
    Driver:   26.21.14.3170
Begin MonoManager ReloadAssembly
- Completed reload, in  0.118 seconds
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
D3D11 device created for Microsoft Media Foundation video decoding.
<RI> Initializing input.

New input system (experimental) initialized
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.

<RI> Initialized touch support.

WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/ParticleLitSoft' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
UnloadTime: 1.048307 ms
WARNING: Shader Unsupported: 'Shader Graphs/VehicleLightsShader' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Controllables\controllable_ChargingStation\manifest.json 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/Mono/data-0000015729E2F340.dll
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Controllables\controllable_TrafficCone\manifest.json 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/Mono/data-00000157517BE030.dll
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
found manifest at Controllables\controllable_TriggerZone\manifest.json 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library C:/Users/lgsvl/Desktop/svlsimulator-windows64-2021.1/simulator_Data/Mono/data-00000157517C17E0.dll
WARNING: Shader Unsupported: 'HDRP/Unlit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
Loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.2087753 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Current Database Version: 7 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Final Database Version: 7 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

[CONN] Instance URL https://wise.svlsimulator.com/
[CONN] Connecting to WISE
Setting up 12 worker threads for Enlighten.
  Thread -> id: 9c90 -> priority: 1 
  Thread -> id: 96b8 -> priority: 1 
  Thread -> id: 78d4 -> priority: 1 
  Thread -> id: c330 -> priority: 1 
  Thread -> id: 1118 -> priority: 1 
  Thread -> id: c134 -> priority: 1 
[... 134 lines cut ...]
  ERROR: SymGetSymFromAddr64, GetLastError: 'Trying to access an invalid address.

' (Address: 00007FFDD704272E)
0x00007FFDD704272E (UnityPlayer) (function-name not available)
  ERROR: SymGetSymFromAddr64, GetLastError: 'Trying to access an invalid address.

' (Address: 00007FFDD704148A)
0x00007FFDD704148A (UnityPlayer) (function-name not available)
  ERROR: SymGetSymFromAddr64, GetLastError: Trying to access an invalid address.

' (Address: 00007FFDD704553C)
0x00007FFDD704553C (UnityPlayer) (function-name not available)
0x00007FFDD704902B (UnityPlayer) UnityMain
  ERROR: SymGetSymFromAddr64, GetLastError: 'Trying to access an invalid address.

' (Address: 00007FF75DB311F2)
0x00007FF75DB311F2 (simulator) (function-name not available)
0x00007FFE2F6784D4 (KERNEL32) BaseThreadInitThunk
0x00007FFE32051821 (ntdll) RtlUserThreadStart

========== END OF STACKTRACE ===========

A crash has been intercepted by the crash handler. For call stack and other details, see the latest crash report generated in:
 * C:/Users/lgsvl/AppData/Local/Temp/LGElectronics/SVLSimulator/Crashes
[CONN] Connected to WISE.

EricBoiseLGSVL, 2021-04-21:
Thanks for logs.  That is a very odd bug.  Navigate to `C:/Users/user/AppData/LocalLow/LGElectronics/SVLSimulator` and delete the folder.  Then link simulator again to your account.

tk-s05, 2021-04-21:
I tried but couldn't start the simulator.
I can launch the simulation in Unity, but I hope I can use the app.
Are there any other possible measures?

Hope for your help, thanks.

EricBoiseLGSVL, 2021-04-22:
Sorry, we don't have many issues running simulator on windows.  Did you delete the appdata folder and run?  Can you please post the player.log as a txt file. it's difficult to parse inline.  Is this a server?  Do you have firewalls blocking?  Can you also check Nvidia control panel and make sure the application is forced to use the GPU?

tk-s05, 2021-04-28:
I tried these...
・LGElectronics delete in appdata folder and run
・Unblocked all firewalls.
・I checked Nvidia control panel and make sure the application is forced to use the GPU

If there is no solution I may have to give up...Thanks

Player.log↓
[Player.log](https://github.com/lgsvl/simulator/files/6388456/Player.log)

EricBoiseLGSVL, 2021-04-28:
Thanks for the log.  Comparing to my log you crash right when you try to connect to the WebUI.  Can you navigate to https://wise.svlsimulator.com ?  What browser are you using?

EricBoiseLGSVL, 2021-04-28:
I feel this could be two things:
1) some firewall/setting is still blocking access to webui
2) You have a default app data folder like this C:\Users\YOURUSERNAME\AppData\LocalLow\LGElectronics\SVLSimulator.  I think your user name is not in English and/or we have a bug that can't write to it or is blocked.  Please try to edit the config file or add one `config.yml` then run again. Be sure to make the SVLDATA folder or you can change this absolute path to what you want, just keep it out of appdata and be sure nothing is blocking letting the app write to your drives.

```
headless: false
read_only: false
api_hostname: "localhost"
api_port: 8181
cloud_url: "https://wise.svlsimulator.com"
data_path: "C:\SVLDATA\LGElectronics\SVLSimulator"
```

closed 2021-07-16 by EricBoiseLGSVL