lgsvl/simulator #1380: car can not move

Anruoqianxi, 2021-04-19:
[image]
[image]
After setting the route, the car is not controlled by Apollo and does not move .  What is the reason?

EricBoiseLGSVL, 2021-04-20:
looks like planning is not active. @lemketron ? @hadiTab ?

EricBoiseLGSVL, 2021-04-23:
Have you made progress on this issue?  This seems like an issue with apollo, have you posted on apollo github issues?

lemketron, 2021-04-23:
> looks like planning is not active

Agree with @EricBoiseLGSVL -- there is a red route, but no blue planning path. Can you show the module controller view in Dreamview? Is Planning turned on? Sometimes after you enable it, you may find that it is disabled if it crashes due to memory issues in the GPU or GPU architecture incompatibility. 

What GPU do you have, how much memory is there, and are you running Apollo and Simulator on the same machine?

closed 2021-07-16 by EricBoiseLGSVL