lgsvl/simulator #1383: Problem for importing new vehicle in the SVL

momala, 2021-04-19:
Hello, 
I am working with the LGSVL 2020.06 with a customize built vehicle. However after upgrading it cant add the vehicle to my library and shows me this error:
Manifest file absent in bundle archive

Should I build my vehicle again in unity with the new setting? it seems that this page is not updated: https://www.svlsimulator.com/docs/archive/2020.06/add-new-ego-vehicle/

ruoxianglee, 2021-04-19:
Hi, I got the similar issue while using LGSLV 2021.01.  I wan to add a new vehicle on the Web UI, so that I can add some sensors needed, and also publish topic "/clock" (actually I am not sure if it is the right way to publish this topic). But finally got a same error just as you mentioned:

[image]

The asset bundle was downloaded here: https://content.lgsvlsimulator.com/vehicles/jaguarxe2015/

ruoxianglee, 2021-04-19:
Hello, I just fixed the problem by following [the instruction](https://www.svlsimulator.com/docs/simulation-content/assets/#build-an-asset) (Adding an Asset & Building an Asset). 

After rebuilding the simulator, I have an asset bundle _vehicle_Jaguar2015XE_ under _build/AssetBundles/Vehicles/_, the files inside which are not same as inside the asset bundle downloaded from https://content.lgsvlsimulator.com/vehicles/jaguarxe2015/. 

Finally, I can add a new vehicle on the Web UI, without the error info: _Manifest file absent in bundle archive_.

I hope that would be useful for you.

EricBoiseLGSVL, 2021-04-20:
Be sure to have linux mono support if installing on windows and windows support if installing on linux. Bundles are cross platform so these modules in Unity hub need to be installed.  Thanks for posting your fix @ruoxianglee , it helps others that have the same issue.

momala, 2021-04-20:
> Hello, I just fixed the problem by following [the instruction](https://www.svlsimulator.com/docs/simulation-content/assets/#build-an-asset) (Adding an Asset & Building an Asset).
>
> After rebuilding [...]

Thank you @ruoxianglee  for helping me and sharing your experience, I will take a look exactly on the document again. But I run the current model in the previous version (2020.06) without any problem. It seems I should consider the new documentation, maybe there is somewhere that I miss it.
@EricBoiseLGSVL  I am using Linux and actually as I said I build my vehicle in unity and used it with your former version. I will check the doc again and post my result here.

momala, 2021-04-21:
Ok, I fixed the problem by building my vehicle package with the new simulator 2021.01 project in the unity 2019.4.18.
I could add the vehicle and publish it in the store. However, I still have a problem starting the localhost:9090 bridge connection for the simulation. But I think that's another problem and is not related to this issue.
Thank you @ruoxianglee and @EricBoiseLGSVL for your supports.

EricBoiseLGSVL, 2021-04-22:
Excellent, glad to hear.  If you can't find an existing open issue with the problem you are having, just open a new issue.  Thanks for posting your solution for others :)

closed 2021-04-22 by EricBoiseLGSVL