lgsvl/simulator #1384: ROS2 Bridge Issue from Rviz after setting up TFTree

sisaha9, 2021-04-19:
This is a followup from #1378

After setting the required map -> odom -> base_link transformations (and sensors linked to base_link) I have the following tf_tree
[image]

Base link to sensor transforms are currently by a static transform publisher (plan to move it to URDF after baseline) while the map -> odom and odom -> base link are done by dual navsat, ekf configurations. I can verify that the /tf is indeed getting transform data published as well

However, Rviz still seems to be giving the errors
[image]

What could be causing this?

EricBoiseLGSVL, 2021-04-20:
Are you converting the data to send to Rviz correctly?

sisaha9, 2021-04-20:
I would guess that's the issue. Currently I just take the received messages and send to robot_localization for generating the TF Tree. I don't have any preprocessing of the sort on the messages (treated it like a Gazebo simulation). Is there somewhere I can read up on to better handle the conversion?

sisaha9, 2021-04-21:
Just realized I didn't specify. But I wasn't trying to view the custom messages defined in lgsvl_msgs. I was referring to standard types like PointCloud2 provided by ROS2 as default

EricBoiseLGSVL, 2021-04-22:
I'm not sure we support all ROS2 types out of the box.  You may need to write the conversion code for ones we don't support.  Right @hadiTab @lemketron ?

sisaha9, 2021-04-22:
I agree. I would probably need to make plugins for the lgsvl_msgs but the main issue was these ROS2 types were not working out of the box

sisaha9, 2021-04-29:
Let me know if I can help in any way

sisaha9, 2021-05-07:
I noticed that the lane following demo given by SVL has a ros2_message_filters package added: https://github.com/lgsvl/lanefollowing/tree/master/ros2_ws/src. Is that something I would need to add? I also found a relevant issue here: https://github.com/ros2/rviz/issues/511 which states that the messages being broadcasted have to have relevant frame information attached to the messages. Is there a way to add that for the SVL?

EricBoiseLGSVL, 2021-05-07:
@daviduhm ?

sisaha9, 2021-05-13:
I think I have found the issue. SVL simulator takes nearly 100% of all 8 of my CPU cores (i7) and only 2GB from the GPU which is causing the ROS transform publishing to lag and inconsistently publish. I tried using the Clock Sensor to circumvent this but it didn't work. This CPU usage seems extremely unnatural though for a Unity simulation. I tried it on a slightly weaker Windows build with an i5 and it ran perfectly fine there. I think this is an issue with the Linux release of the simulator

EricBoiseLGSVL, 2021-05-13:
Yes, windows is always more performant.  Next release uses a new version of Unity that is much better on Linux.

sisaha9, 2021-05-13:
Great! Will wait for that. Closing this issue for now then

closed 2021-05-13 by sisaha9

sisaha9, 2021-05-15:
@EricBoiseLGSVL Just wanted to know but when would the next release be out approximately?

EricBoiseLGSVL, 2021-05-17:
Next release is being tested now but it is small. We will have a larger update soon after