lgsvl/simulator #1385: How to get the ground truth of traffic road lines, parking lines. etc.?

zengtaiping, 2021-04-20:
I want to build a map with road lines. How to get the ground truth of traffic road lines, parking lines. etc.?
The Segmentation Camera Sensor only shows the information about the ground, tree, and grassland, like the following:
[image]

EricBoiseLGSVL, 2021-04-20:
This map is not open source so you cannot create the lane/line meshes.
In your own map, you can open Simulator -> Detect Lane Lines.
See documentation [here](https://www.svlsimulator.com/docs/simulation-content/lane-line-detector/)

zengtaiping, 2021-04-21:
Thank you so much for your explanation @EricBoiseLGSVL .

closed 2021-04-21 by zengtaiping

EricBoiseLGSVL, 2021-04-21:
@zengtaiping no problem and glad to help