lgsvl/simulator #1386: Adding a fixed sensor that publishes ROS topics

risk-v, 2021-04-20:
Hi SVL team,

I would like to setup a fixed camera sensor that is detached from the car and publishes a ROS topic e.g. "/simulator/fixed_camera". Is it possible to achieve this functionality using the latest version of the simulator? Thanks in advance.

Best regards

EricBoiseLGSVL, 2021-04-20:
Next release we will have this feature :)

risk-v, 2021-04-21:
Hi Eric,

Thanks for your reply. That's brilliant.  Is it going to be released in the near future? In the meanwhile, I've trying to create a fixed sensor inside Unity with the latest SVL Simulator (2021 version) running. I managed to duplicate the sensor (e.g. main camera attached to the car), unparent it and change the topic name but that didn't work, i.e. running "rostopic list" didn't show an updated list of topics. If that works it could be an alternative way of dealing with fixed sensors till the next release :) Or is there a better way of doing this manually? Thanks again Eric really appreciate it.

Regards

EricBoiseLGSVL, 2021-04-22:
No problem, glad to help.

So what we have now that will release soonish.  We simplified egos so the only component required is a rigidbody and agent controller.

In the meantime, you can take existing vehicle but turn off mesh renderers, make rigidbody kinematic and move the spawn point to a specific point or use api to place in the position you want. You may need to edit source for unexpected errors.

If you don't have access to map but you can edit the simulator source, you could parent the sensor `after` it is parented to the vehicle.  You can't copy because of the bridge setup logic.

Hope this helps

risk-v, 2021-04-27:
Hi Eric,

Thanks again for your reply. I have it working now :) I had to make a few changes in the code too.

Regards

closed 2021-04-27 by EricBoiseLGSVL