lgsvl/simulator #1387: About custom map download

qiongwu1003, 2021-04-21:
Hi, SVL team

In the 202101 SVL simulator, I add a custom map to the cloud. When I run the simulations I need to download the map first. When the map is large or the network is slow, the download is very slow and aften fails.
Can I put the custom map file in the locally in advance? Then the simulator will run directly and over the download step.

Thanks.

EricBoiseLGSVL, 2021-04-21:
@qiongwu1003 sorry we are in the process of adding the feature for resume downloads.  We hope to have it done soon.  You can use developer settings in the simulator settings but it is still work in progress.  Open the scene in Unity editor, open development settings, select a local vehicle, paste json sensor config in the settings, run from scene in editor.  It has bugs but it will let you test until we can implement resume download.

qiongwu1003, 2021-04-22:
@EricBoiseLGSVL Thank you very much for your reply, it is of great help to me. And I also look forward to the feature for resume downloads.

closed 2021-04-22 by qiongwu1003