lgsvl/simulator #1388: ConnectionRefusedError

Anruoqianxi, 2021-04-22:
[image]
when I use PythonAPI connect to 10.28.154.31:9090,it refuse to connect.10.28.154.31 is the IP of computer which is running Apollo. bridge.sh is already running.The computer does not have any anti-virus software installed.What is the reason?

Anruoqianxi, 2021-04-22:
[image]

EricBoiseLGSVL, 2021-04-23:
do you have any other application using this port/address?  Check the applications running and their ports

lemketron, 2021-04-23:
Try using netstat to look for open ports. Run this before starting the Apollo bridge to see if some other service is listening on port 9090, and then run it again after starting the bridge.
https://www.eukhost.com/forums/forum/technical-support/tutorials-how-tos/19396-how-to-find-open-ports-on-a-windows-server

You should be able to connect to the bridge using `telnet localhost 9090` from the Linux machine on which you are running Apollo, and also from another machine on the same local network using `telnet 10.28.154.31 9090`. If it doesn't work then there may be some issue on your apollo machine; you might research or ask about this on the [Apollo issues page](https://github.com/apolloauto/apollo/issues) page.

I just tested this on my system:
```
# before starting bridge.sh in apollo docker:

$ telnet 192.168.1.164 9090
Trying 192.168.1.164...
telnet: connect to address 192.168.1.164: Connection refused
telnet: Unable to connect to remote host

# after starting bridge.sh in apollo docker:

$ telnet 192.168.1.164 9090
Trying 192.168.1.164...
Connected to 192.168.1.164.
Escape character is '^]'.
blah
Connection closed by foreign host.
```

closed 2021-07-16 by EricBoiseLGSVL