lgsvl/simulator #1389: The relationship between apollo5 map and base_map

yumianhuli2, 2021-04-22:
Hello！
What is the relationship  between apollo5 hd map and base_map.bin/base_map.txt?

If I want the Apollo algorithm to run on the new map, how do I get base_map.txt or base_map.bin for the new map?

Thank you！

lemketron, 2021-04-22:
You can read more about the Apollo map formats here:
https://github.com/ApolloAuto/apollo/tree/master/modules/map/data#difference-between-base_map-routing_map-and-sim_map

SVL Simulator can generate maps in Apollo or OpenDrive format; they are not the same. Please be sure you are using the correct file for Apollo.

yumianhuli2, 2021-04-23:
> You can read more about the Apollo map formats here:
> https://github.com/ApolloAuto/apollo/tree/master/modules/map/data#difference-between-base_map-routing_map-and-sim_map
>
> SVL Simulator can generate [...]

 Yes,I mean how to  generate this base_map.txt or base_map.bin for own maps in lgsvl simulator?

Thank you!

lemketron, 2021-04-23:
> Yes,I mean how to generate this base_map.txt or base_map.bin for own maps in lgsvl simulator?

Are you talking about building your own environment in the simulator? Did you also provide [annotations](https://www.svlsimulator.com/docs/simulation-content/map-annotation/)?

If so, then you can export an Apollo map as documented here:
https://www.svlsimulator.com/docs/simulation-content/map-annotation/#export-map-annotations

That will export a base_map.bin that you can install into Apollo as documented here:
https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/#adding-an-hd-map

I don't know about adding maps to the planning directory; they go in `/apollo/modules/map/data`.

You can find out about the apollo map conversion tools used to generate other apollo (routing and sim) maps here:
https://github.com/ApolloAuto/apollo/tree/master/modules/map/data#difference-between-base_map-routing_map-and-sim_map

yumianhuli2, 2021-04-28:
> Are you talking about building your own environment in the simulator? Did you also provide [annotations](https://www.svlsimulator.com/docs/simulation-content/map-annotation/)?
>
> If so, then you can [...]

OK，Thank you！By the way,what is the version of opendrive in lgsim?

lemketron, 2021-04-28:
> OK，Thank you！By the way,what is the version of opendrive in lgsim?

SVL Simulator currently supports version 1.4 of OpenDrive:
https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L99

yumianhuli2, 2021-04-29:
> > OK，Thank you！By the way,what is the version of opendrive in lgsim?
> 
> SVL Simulator currently supports (importing and exporting) version 1.4 of OpenDrive:
> https://github.com/lgsvl/simulator/blob/8b02c852c0642347f0b05125c726ceb09bfa0b26/Assets/Scripts/Editor/OpenDriveMapImporter.cs#L99

ok，got it！

Thinsheep, 2021-05-03:
When I import .xord Open Drive map file from LGSVL to my software I only see the layout of the map, not visual data graphic like, buildings, trees, 3D data of the map like Borregas Avenue and San Francisco. Does that 3D visual graphic come from OpenSceneGraph or something like that? How is it in LGSVL?