lgsvl/simulator #1390: 【LGSVL Unity Install Error】

shandaolu, 2021-04-22:
Hello, I met the question when I install the LGSVL and Unity. According to the instruction (https://www.svlsimulator.com/docs/installation-guide/build-instructions/), when i import the project，it comes out that" your project was last saved with a different version of unity.  .... ... The saved project (before 5.0 ) does not match the launched editor(2019.4.18f1) ...... " But I haved installed the Unity 2019.4.18f1. 

My operating system is Win 10. The version of Unity Hub is 2.4.3. The version of Unity is 2019.4.18f1. The simulator is master. Please  help me solve this problem. Thankyou

EricBoiseLGSVL, 2021-04-22:
be sure you are on the release tag, not master

shandaolu, 2021-04-23:
Thank you for your reply.
I execute through this instruction:
git clone https://github.com/lgsvl/simulator.git. Which release tag do I choose?

EricBoiseLGSVL, 2021-04-23:
2021.1

closed 2021-07-16 by EricBoiseLGSVL