lgsvl/simulator #1391: Map is missing annotation MapHolder and child holders!

cxf99, 2021-04-23:
Map is missing annotation MapHolder and child holders! Please add to scene and set holder transforms
UnityEngine.Debug:LogError(Object)
Simulator.Map.MapManagerData:.ctor() (at Assets/Scripts/Map/MapManagerData.cs:25)
Simulator.Editor.MapMeshes.MapMeshBuilder:BuildMesh(GameObject, MapMeshMaterials, Boolean, LaneLineOverride) (at Assets/Scripts/Editor/Map/Meshes/MapMeshBuilder.cs:266)

cxf99, 2021-04-23:
@lemketron Sorry, how to solve this problem，thank you，How to  annotation MapHolder and child holders？

EricBoiseLGSVL, 2021-04-23:
If you create a new scene, we include this automatically.  If you already have a scene, you can manually create this.
Please see CubeTownMap as an example on how this is setup.

[image]
[image]
[image]
[image]
[image]

Please look at the annotation documentation [here](https://www.svlsimulator.com/docs/simulation-content/map-annotation/).

closed 2021-07-16 by EricBoiseLGSVL