lgsvl/simulator #1393: How to get the exact coordinates of waypoints in the simulator?

GabrielZZZ, 2021-04-26:
Hi guys,

I am new to use the python script of the simulator. Can anyone tell me how to get the exact position of waypoints on the map to put in the script? For instance, I want to place a car at a corner in the map of Borregas Avenue, but I do not know the exact position on the simulator so the function call `waypoints.appeend` cannot be implemented.

I have seen the documentation and example codes but I did not find content about how to get the position of waypoints.

Thanks a lot!

GabrielZZZ, 2021-04-26:
Plus: I noticed the visual editor does not work with Apollo 6.0 (according to documentation). Therefore, except the ones listed in the previous issues (e.g. "drive" the car to the location and write down the coordinate), are there better ways to do this? I wonder how the author of the existing code found out the coordinate (relative coordiates)

EricBoiseLGSVL, 2021-04-26:
If you open the simulator in Editor, you can navigate while the simulation is running.  Start with the LoaderScene then run the simulation.  Then you can add an empty object and drag it to any point in the map.  You can also create a sensor that outputs positional data.  There are a few sensors that report orientation but it is in right hand coordinates, you could convert to unity left hand and then you wouldn't need any changes.

GabrielZZZ, 2021-04-27:
Thank you @EricBoiseLGSVL . I will look in the way you mentioned. Besides, I have found the function `sim.map_from_gps` in the python-API samples that can convert GPS to map coordinates in vector form. I think it is possible to use it with the API simulation that when I drive the Ego Car, the console outputs the map coordinates simultaneously.

GabrielZZZ, 2021-04-28:
Hi. There is one point I am still confused that in the sample codes offered [here](https://www.svlsimulator.com/docs/running-simulations/python-runtime/#reading-logs), how is the position of the NPC vehicles confirmed, i.e. how to obtain this number (**50** * forward)?
[image]
Is it through guessing and trying time by time?

EricBoiseLGSVL, 2021-04-28:
This is just some vector math.
```
ego_state.transform.position = spawns[0] // spawn position
forward = transform_to_forward(spawns[0]) // get the forward rot vector of the first spawn position
ego_state.transform.position = ego_state.transform.position + (50 * forward) // multiply ego pos forward vector by 50 and add to ego position.  You can print these positions in the code to see.
```

Does this help?

GabrielZZZ, 2021-04-29:
Yes, thank you. My confuse is mainly here:

> `ego_state.transform.position = ego_state.transform.position + (50 * forward) // multiply ego pos forward vector by 50 and add to ego position.  You can print these positions in the code to see.`

How to ensure the number is **50** rather than others? Is it obtained by repeatedly experimenting, e.g. printing the positions in the code to adjust?

duongtran11, 2021-05-12:
The spawn[0] is the SpawnInfo object you see in the hierarchy in Unity. so you can drag it any where you want in your map. It just like any other object. if you create an other SpawnInfo (by creating a 3D object -> add Spawn Info (Script) to it -> change it name to SpawnInfo2 (not nessessary), now you can use a spawn[1] object in your code. 
I used to create up to 4 SpawnInfo object to spawn NPC vehicles in 4 different places in the map

marcusvinicius178, 2021-09-24:
Hi @EricBoiseLGSVL I need to get the EGO vehicle Position + Orientation in a vector way:

The pyton API script 21 gives me the Transform from the spawn position:
tr = spawns[0]
Then it transforms this vector in GPS coordinates:
gps = sim.map_to_gps(tr)
And convert later from gps to transform coordinates
t = sim.map_from_gps(norhing=gps.northing,easting = gps.easting, altitude= gps.altitude, orientation=gps.orientation)

Ok but I do not need the initial spawn position (spawns[0]). I need to get the vehicle gps data after I move with the EGOvehicle to other position that I wish  ( I am controlling with the keyboard).

Then when the EGO be located in this position I wish to get the gps data from this position for later convert this gps data into transform data.

I need this because I need to send the navigation goal for the EGO vehicle in a standard and fixed spot, to compare other path planners. I will send the position by command line, and not by rviz icon. That is why I need the exact location of the goal spot in terms of latitude and longitude...but I do not know how to get it using PythonAPI
I would like to move the car ang see in my console printing the car position. I have tried this code:

`from environs import Env
import copy
import lgsvl
import time
import os

print("NPC waiting EGO gets closer")
env = Env()

LGSVL__SIMULATOR_HOST = env.str("LGSVL__SIMULATOR_HOST", "127.0.0.1")
LGSVL__SIMULATOR_PORT = env.int("LGSVL__SIMULATOR_PORT", 8181)
LGSVL__AUTOPILOT_0_HOST = env.str("LGSVL__AUTOPILOT_0_HOST", "127.0.0.1")
LGSVL__AUTOPILOT_0_PORT = env.int("LGSVL__AUTOPILOT_0_PORT", 9090)

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"),8181)

#MAP
if sim.current_scene == "CubeTown":
    sim.reset()
else:
    sim.load("CubeTown")

spawns = sim.get_spawn()

state = lgsvl.AgentState()
state.transform = spawns[0]

forward = lgsvl.utils.transform_to_forward(spawns[0])

# Agents can be spawned with a velocity. Default is to spawn with 0 velocity

ego = sim.add_agent(env.str("LGSVL__VEHICLE_0", lgsvl.wise.DefaultAssets.ego_lincoln2017mkz_apollo5), lgsvl.AgentType.EGO, state)

#DATA 1

print("\n DATA 1")
print("\n Current time 1 = ", sim.current_time)
print("Current frame 1 = ", sim.current_frame)

tr1 = spawns[0]

print("\n Default transform 1: \n {}".format(tr1))

# There are functions to convert to and from GPS coordingates
# This function takes position and rotation vectors and outputs the equivalent GPS coordinates
gps1 = sim.map_to_gps(tr1)
print("\n GPS coordinates 1: \n {}".format(gps1))

# This function can take either lat/long or northing/easting pairs as inputs and will provide equivalent position and rotation vectors
# Altitude and orientation are optional
t1 = sim.map_from_gps(
    northing=gps1.northing,
    easting=gps1.easting,
    altitude=gps1.altitude,
    orientation=gps1.orientation,
)
print("\n Transform from northing/easting 1: {}".format(t1))

t2 = sim.map_from_gps(latitude=gps1.latitude, longitude=gps1.longitude)
print("\n Transform from lat/long without altitude/orientation 1: {}".format(t2))

sim.run(time_limit=15)

input("\nPress Enter to continue driving for 20 seconds ")

#DATA 2

print("\n DATA 2")
print("\n Current time 2 = ", sim.current_time)
print("Current frame 2 = ", sim.current_frame)

#data = gps_sensor.data()
#latitude  = data.latitude
#longitude = data.longitude

spawns2 = sim.get_spawn()
tr2 = spawns2
print("\n Default transform 2: \n {}".format(tr2))

# There are functions to convert to and from GPS coordingates
# This function takes position and rotation vectors and outputs the equivalent GPS coordinates
gps2 = sim.map_to_gps(tr2)
print("\n GPS coordinates 2: \n {}".format(gps2))

# This function can take either lat/long or northing/easting pairs as inputs and will provide equivalent position and rotation vectors
# Altitude and orientation are optional
t3 = sim.map_from_gps(
    northing=gps2.northing,
    easting=gps2.easting,
    altitude=gps2.altitude,
    orientation=gps2.orientation,
)
print("\n Transform from northing/easting 2: {}".format(t3))

t4 = sim.map_from_gps(latitude=gps2.latitude, longitude=gps2.longitude)
print("\n Transform from lat/long without altitude/orientation 2: {}".format(t4))

sim.run(time_limit=20)

`

But even if I move the car it prints me the same position, does not get the new position the car is located. Some idea? please?