lgsvl/simulator #1395: Is it OK apollo 5.0(fully analysis) scenario with apollo 6.0 in server?

powerpzt, 2021-04-27:
Hello.

I installed apollo 6.0 in my server.
However, LGSVL supports apollo 5.0(fully analysis) and apollo 6.0(modular test).
Because my research is related to perception, I need to use apollo 5.0(fully analysis).

Do I need to install another apollo 5.0 in my server?
Or, does the apollo 6.0 support the apollo 5.0(fully analysis)?

Actually, I tested apollo 6.0 with 5.0(fully analysis), and the result looks like that it works.
However, I am afraid that there would be collision during research.

EricBoiseLGSVL, 2021-04-27:
You can use 6.0 but apollo has issues with perception at the moment.  We provide the title as an example of the sensors needed for the apollo version but you can use what you want, using other sensors should still work as long as apollo gets the data it needs.  So if you are needing perception to work, I recommend using 5.0 until perception is fixed.  @lemketron @hadiTab

lemketron, 2021-04-27:
As @EricBoiseLGSVL mentioned, the different configurations are just different bundles of sensors. Modular Testing configurations were developed to omit LiDAR and camera sensors from the vehicle and instead substitute the simulator ground truth information so you don't need to run Apollo perception or traffic light modules.

This was useful for two reasons:

- First, running those Apollo modules in addition to the prediction module consume a lot of GPU memory and if you're doing that on a single machine with the SVL Simulator also sharing the GPU, it is easy to run out of GPU memory or CPU processing power. So by using modular testing it is easier to run Apollo and simulator on a single machine.
- Second, Apollo 6.0 (and master) went through a few GPU code refactors as the project gained support for Nvidia 20x0 and 30x0 series GPUs that left perception unavailable for a while.

If perception is working for you on your GPU with Apollo 6.0 or master, and you have enough computing power to run Apollo with perception along side of SVL Simulator (preferably on two separate machines) then feel free to use it.

The primary difference in the "Apollo 6.0" sensor configuration is that it includes the clock sensor to support "stepped simulation" which is not available in Apollo 5.0 -- though to be fair, having that clock sensor in Apollo 5.0 shouldn't hurt anything but it might be confusing since Apollo 5.0 does not include the changes necessary to support using the simulation clock. Note that the clock sensor does nothing without both the support in Apollo AND a simple change to `cyber.pb.conf` to change the `clock_mode` as described in the [clock sensor documentation](https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock). 

You can of course make your own sensor configurations but we encourage users to use the standard configurations since those are the ones we test with:

- **5.0 (full analysis)** = sensors needed for **Apollo 5.0 or 6.0** _using Apollo perception_, plus extra sensors to generate date for test reports (e.g. video recording sensor, stop line sensor, vehicle odometry sensor, and should include comfort sensor as well).
- **5.0 (modular testing)** = sensors needed for **Apollo 5.0** _using simulator ground truth instead of Apollo perception_
- **6.0 (modular testing)** = sensors needed for **Apollo 6.0** _using simulator clock sensor and ground truth instead of Apollo perception_

Hope this helps to understand the different sensor configurations.