lgsvl/simulator #1397: LGSVL Python API (Dreamview) cannot start Apollo modules

ebadi, 2021-04-27:
Hello,
I am using the following code to communicate with Apollo master via Python API:
```
agents["ego"].connect_bridge("127.0.0.1", 9090)
dv = lgsvl.dreamview.Connection(sim, agents["ego"], "127.0.0.1")
dv.set_hd_map('borregas_ave')
dv.set_vehicle('Lincoln2017MKZ_LGSVL')

modules = [
        'Localization',
        # 'Perception',
        'Transform',
        'Routing',
        'Prediction',
        'Planning',
        # 'Traffic Light',
        'Control'
    ]
destination = spawns[0].position + 135 * forward
dv.setup_apollo(destination.x, destination.z, modules)

```
But I am receiving the following error messages: 

```
INFO:lgsvl.dreamview.dreamview:Starting Localization module...
INFO:lgsvl.dreamview.dreamview:Starting Transform module...
INFO:lgsvl.dreamview.dreamview:Starting Routing module...
INFO:lgsvl.dreamview.dreamview:Starting Prediction module...
INFO:lgsvl.dreamview.dreamview:Starting Planning module...
INFO:lgsvl.dreamview.dreamview:Starting Control module...
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Control is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Localization is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Planning is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Prediction is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Routing is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Transform is not running!!!
INFO:lgsvl.dreamview.dreamview:10 seconds have passed but Ego hasn't received any control messages.
INFO:lgsvl.dreamview.dreamview:Please also check if your route has been set correctly in Dreamview.
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Control is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Localization is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Planning is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Prediction is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Routing is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Transform is not running!!!
INFO:lgsvl.dreamview.dreamview:20 seconds have passed but Ego hasn't received any control messages.
INFO:lgsvl.dreamview.dreamview:Please also check if your route has been set correctly in Dreamview.
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Control is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Localization is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Planning is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Prediction is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Routing is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Transform is not running!!!
INFO:lgsvl.dreamview.dreamview:30 seconds have passed but Ego hasn't received any control messages.
INFO:lgsvl.dreamview.dreamview:Please also check if your route has been set correctly in Dreamview.
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Control is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Localization is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Planning is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Prediction is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Routing is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Transform is not running!!!
INFO:lgsvl.dreamview.dreamview:40 seconds have passed but Ego hasn't received any control messages.
INFO:lgsvl.dreamview.dreamview:Please also check if your route has been set correctly in Dreamview.
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Control is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Localization is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Planning is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Prediction is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Routing is not running!!!
WARNING:lgsvl.dreamview.dreamview:Warning: Apollo module Transform is not running!!!
INFO:lgsvl.dreamview.dreamview:50 seconds have passed but Ego hasn't received any control messages.
INFO:lgsvl.dreamview.dreamview:Please also check if your route has been set correctly in Dreamview.
ERROR:lgsvl.dreamview.dreamview:No control message from Apollo within 60.0 seconds. Aborting...
Traceback (most recent call last):
  File "02-PerformVehicleFollowing.py", line 90, in <module>
    dv.setup_apollo(destination.x, destination.z, modules)
  File "/home/wave/.local/lib/python3.6/site-packages/lgsvl/dreamview/dreamview.py", line 338, in setup_apollo
    raise WaitApolloError()
lgsvl.dreamview.dreamview.WaitApolloError
```

### Setup
- lgsvlsimulator-linux64-2020.06
- Python API : 2021.1 (tag 2020.06 does not support dreamview)
- Apollo (master): (see below)

```
wave@wave:~/repositories/apollo$ bash docker/scripts/dev_into.sh
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

[wave@in-dev-docker:/apollo]$ ./scripts/bootstrap_lgsvl.sh
Starting modules/drivers/tools/image_decompress/launch/image_decompress.launch
nohup: appending output to 'nohup.out'
[ OK ] Launched module monitor.
[ OK ] Launched module dreamview.
Dreamview is running at http://localhost:8888
[wave@in-dev-docker:/apollo]$ ./scripts/bridge.sh
```
 Cyber monitor:

[image]

EricBoiseLGSVL, 2021-04-27:
If you are using python 2021.1 tag you need to use simulator release 2021.1.  Apollo master requires modular testing because apollo perception is bugged.  Please update to the latest simulator release for these features.

ebadi, 2021-04-27:
Thanks @EricBoiseLGSVL for the answer.  Unfortunately, for IEEE challenge (http://av-test-challenge.org/deliverables.html) we were initially requested ( @lemketron)  to use the 2020.1 version. I am really eager to try the newer version but I am not sure if it can potentially break the configuration or if the maps/vehicle assets are compatible, etc. It is not ideal to break my setup just few days before the deadline. 
Is there a version of LGSVL Python API that is compatible with 2020.06 and has the dreamview APIs?

ebadi, 2021-04-27:
I tried **SVL 2021.1** python API and simulator, none of the NHTSA test work since Apollo misses these maps:
```
PythonAPI-LGSVL/examples/NHTSA/Vehicle-Following$ python3 VF_S_65_Slow.py 
[ ERROR] [lgsvl.dreamview.dreamview] HD Map Straight1LaneSame was not set. Verify the files exist in /apollo/modules/map/data/Straight1LaneSame and restart Dreamview -- Aborting..
```

I tried my own code with the same setup and got the following error message: 
```
    agents["ego"].connect_bridge("127.0.0.1", 9090)
  File "/home/wave/.local/lib/python3.6/site-packages/lgsvl/utils.py", line 23, in new_f
    return f(*args, **kwargs)
  File "/home/wave/.local/lib/python3.6/site-packages/lgsvl/agent.py", line 174, in connect_bridge
    {"uid": self.uid, "address": address, "port": port},
  File "/home/wave/.local/lib/python3.6/site-packages/lgsvl/remote.py", line 69, in command
    raise Exception(data["error"])
Exception: Agent '7fe3fbb3-8485-47f6-a3dd-3a082cea134b' is missing bridge client
```