lgsvl/simulator #1399: Missing NHTSA maps

ebadi, 2021-04-28:
Apollo : master
SVLsimulator : svlsimulator-linux64-2021.1
Python API: 2021.1 and master

```
/examples$ python3 NHTSA/Vehicle-Following/VF_S_45_Slow.py
[ ERROR] [lgsvl.dreamview.dreamview] HD Map Straight1LaneSame was not set. Verify the files exist in /apollo/modules/map/data/Straight1LaneSame and restart Dreamview -- Aborting...
```

I tried to download apollo50 map(base_map.bin) from wise.svlsimulator.com and place it in the specified folder with no success.

revati-naik-lgsvl, 2021-04-29:
Hi @ebadi, could you please confirm the folder name for the map? Make sure it is renamed as "Straight1LaneSame". The path to the map should look something like this "apollo-5.0/modules/map/data/Straight1LaneSame/base_map.bin".

ebadi, 2021-04-29:
[image]

[image]

[image]

[image]

[image]

Now I get another error. Do you have any idea what has been wrong ?

lemketron, 2021-04-29:
It looks like you launched a simulation (maybe random traffic simulation) in interactive mode and pressed the "play" triangle button?

You need to launch an API-Only simulation to run Python scripts.

https://www.svlsimulator.com/docs/running-simulations/running-simulator/#apionlysimulation

ebadi, 2021-04-29:
Thanks@lemketron I actually used the API only mode: 
[image]

lemketron, 2021-04-29:
Can you post the Player.log file then? (Read through it first and see if you spot any clues.) It will be shorter if you quit and restart the simulator before testing the script.

ebadi, 2021-04-29:
I took a look at it but I see no problem with it. 

[Player.TXT](https://github.com/lgsvl/simulator/files/6401587/Player.TXT)

ebadi, 2021-04-30:
I restarted my PC several times yesterday and it didn't work. It is now working today and I have no idea why! I close this issue for now. 
Thanks for your help.

ebadi, 2021-04-30:
I found out the issue.
Apollo crashes when an invalid map is given by `dv.set_hd_map('mapname')`. I am still struggling to find out what the problem is with the map.

Yuchen413, 2021-06-09:
> I found out the issue.
> Apollo crashes when an invalid map is given by `dv.set_hd_map('mapname')`. This means Apollo, the bridge and svl have to be restarted.
> 
> I am still struggling to find out what the problem is with the map.

The map folder name in Apollo is "apollo-5.0/modules/map/data/straight1lanesame" (All lower case). Copy the map folder and rename it as Straight1LaneSame (with uppercase letter) solved my problem.

revati-naik-lgsvl, 2021-06-09:
@ebadi Please rename your map folder name to match with the map name shown in the LGSVL map store/map library.

ebadi, 2021-06-10:
Thanks, I will try this and come to you. I close the issue for now.

closed 2021-06-10 by ebadi