lgsvl/simulator #1402: How can remotely control Simulator use Python API runing on another machine?

hanyh1191, 2021-04-30:
I want to run Python API on Ubuntu, and run simulator on windows use PythonAPI mode, After the Simulator got readly,  And I run this in Ubuntu terminal:
>>>import lgsvl
>>>sim = lgsvl.Simulator(address = "10.128.XXX.XXX", port = 8181) , address is the windows IP.

Then I get Error:

**Exception in thread Thread-9:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/hanyh/lgsvl/lgsvl_python/PythonAPI/lgsvl/remote.py", line 29, in run
    self.loop.run_until_complete(self.process())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "/home/hanyh/lgsvl/lgsvl_python/PythonAPI/lgsvl/remote.py", line 37, in process
    self.websocket = await websockets.connect(self.endpoint, compression=None)
  File "/home/hanyh/.local/lib/python3.6/site-packages/websockets/py35/client.py", line 12, in __await_impl__
    transport, protocol = await self._creating_connection
  File "/usr/lib/python3.6/asyncio/base_events.py", line 794, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.6/asyncio/base_events.py", line 781, in create_connection
    yield from self.sock_connect(sock, address)
  File "/usr/lib/python3.6/asyncio/selector_events.py", line 439, in sock_connect
    return (yield from fut)
  File "/usr/lib/python3.6/asyncio/selector_events.py", line 469, in _sock_connect_cb
    raise OSError(err, 'Connect call failed %s' % (address,))
ConnectionRefusedError: [Errno 111] Connect call failed ('10.128.XXX.XXX', 8181)**

Why? Is we can not do this? Can't run python API on another machine?
Thinks.

EricBoiseLGSVL, 2021-05-01:
Why can't you run the python api on the widows machine?

hanyh1191, 2021-05-02:
@EricBoiseLGSVL  hello, I first run python API  and Simulator on the windows machine at the same time. I run Simulator with "API only mode", and then run this in windows terminal:
`import lgsvl
sim = lgsvl.Simulator(address = "localhost", port = 8181)`
About ten seconds later,   
**Unexpected exception in keepalive ping task
Traceback (most recent call last):
  File "C:\Users\hanyh-matebook14\AppData\Roaming\Python\Python39\site-packages\websockets\protocol.py", line 984, in keepalive_ping
    ping_waiter = yield from self.ping()
  File "C:\Users\hanyh-matebook14\AppData\Roaming\Python\Python39\site-packages\websockets\protocol.py", line 598, in ping
    yield from self.write_frame(True, OP_PING, data)
  File "C:\Users\hanyh-matebook14\AppData\Roaming\Python\Python39\site-packages\websockets\protocol.py", line 911, in write_frame
    with (yield from self._drain_lock):
TypeError: 'Lock' object is not iterable**
This Exxception also happend in pythonAPI/Quickstart,  but will igonre the Exception and continue run simu. Why do exception arise?
OK, latter I use two machine to run Simulator on Windows and run python API on Linux,  and I found can not connect, just like my origin question,  I also want to know Why?
Thanks.

EricBoiseLGSVL, 2021-05-03:
If you can't connect on the same machine, you won't be able to connect with two.  What version of python api are you on?  I would try to reinstall python in case you are not getting latest, even if the repo is on the right version.  Right @lemketron @hadiTab ?

hanyh1191, 2021-05-04:
@EricBoiseLGSVL  Thanks, After my Python version was reduced from 3.9 to 3.6, it is success on the same machine. But, I only can use the "localhost" or "127.0.0.1" to Connect Simulator, I see the doc write "**You can specify a different address as hostname or IP address.** " then I use "sim = lgsvl.Simulator(address = "10.128.xxx.xx", port = 8181)" it raise an Error "**ConnectionRefusedError: [Errno 111] Connect call failed ('10.128.xxx.xx', 8181)**".
What should I do? Is only can use the localhost?

EricBoiseLGSVL, 2021-05-04:
do you have another application using that port already?

hanyh1191, 2021-05-07:
@EricBoiseLGSVL  Thanks, maybe not, I use "localhost" or "127.0.0.1" can connect successful.
`sim = lgsvl.Simulator(address = "localhost", port = 8181)`
but I use the IP address of the same machine, It failed.
`sim = lgsvl.Simulator(address = "10.128.xxx.xx", port = 8181)`
I think it is not  port,   is IP address can not be used but only localhost can use?
And if others can use the IP address, maybe is my problem.

EricBoiseLGSVL, 2021-05-07:
@hadiTab @lemketron ?