lgsvl/simulator #1403: Does svlsimulator.com support API to start the simulation?

left4taco, 2021-04-30:
We are developing a system based on ROS2, and LGSVL would be a very good tool to test out application. However, occasionally some code change breaks the simulator/bridge/our application (either on our side or code change from lgsvl, doesn't matter). So we are planning to add an integration test in our CI to verify that our code works well with LGSVL.

The idea is to start LGSVL in headless more, start the [ros2-lgsvl-bridge](https://github.com/lgsvl/ros2-lgsvl-bridge), then verify that we can receive data from lgsvl topics(e.g., `/lidar_front/points_raw`)

Currently here's what I do to start LGSVL:
1. Download 2021.1 tarball, install it somewhere. 
2. Clone `ros2-lgsvl-bridge`, compile and start. It'll listen to port 9090
3. Start LGSVL, login to `svlsimulator.com`, configure the cluster
4. Start the simulation I want to run in the webpage
5. Monitor the topic by `ros2 topic echo /lidar_front/points_raw`  to verify I got information from LGSVL

I can automate the first 3 steps, i.e., write script to run them in CI. However, the obstacle is how to start the simulation without interaction to the webpage. 

Is there any hidden API that I can take advantage of? Didn't find any in the website. 

Thanks in advance.

t0ny-peng, 2021-07-02:
@EricBoiseLGSVL Hi Eric! I heard from Hadi that the tool will be release soon. Any update on that? Thanks!

EricBoiseLGSVL, 2021-07-02:
We are working on this now and hope to release soon.  I don't think it will make 2021.2 release.

lemketron, 2021-07-02:
We are preparing a command line interface to configure and control simulations in the same manner as interacting with the wise.simulator.com web interface. We do not yet have a schedule for the release but hope to have something available later this year (and may make it available in preview form to those with an immediate/business need).

If this is something you'd be interested in, please share your interest and use case with us using the "Contact Us" (or "Request Demo") form at https://www.svlsimulator.com/

lemketron, 2022-02-05:
@t0ny-peng wrote:

> Currently here's what I do to start LGSVL simulator:
> 
> 1. Download 2021.1 tarball, install it somewhere.
> 2. Clone `ros2-lgsvl-bridge`, compile and start. It'll listen to port 9090
> 3. Start LGSVL simulator, login to `svlsimulator.com`, configure the cluster
> 4. Start the simulation I want to run in the webpage
> 5. Monitor the topic by `ros2 topic echo /lidar_front/points_raw`  to verify I got information from LGSVL simulator(via ros2-lgsvl-bridge)
> 
> I can automate the first 3 steps, i.e., write script to run them in CI. However, the obstacle is how to start the simulation without interaction to the webpage.
> 
> Is there any hidden API that I can take advantage of? Didn't find any in the website.

We did in fact build, release, and document a tool for local automation including starting the simulation (step 4 in your post above).

You can read more about the local automation tool here:
https://www.svlsimulator.com/docs/running-simulations/local-automation/

There's a related tutorial on setting it up for CI/CD here:
https://www.svlsimulator.com/docs/tutorials/local-automation-tutorial/

It's worth pointing out that the local automation tool is intended to communicate with the SVL web user interface (aka "WISE") and does not directly start and stop local instances of the simulator nor remove the need to register a simulator instance with an account and create a cluster, simulation, etc.

Due to the [SVL Simulator Sunset](https://www.svlsimulator.com/news/2022-01-20-svl-simulator-sunset/) announcement, the above may or may not be useful to you but it should work as long as the WISE server is still available.

The other option for those who wish to continue using SVL Simulator (and I believe several folks may be working on this) would be for someone to release changes to allow SVL Simulator to start up from the command line in "Python API" (or interactive) mode and load assets locally thus removing the dependency on the WISE server being available. At that point, local (CI/CD and other) testing could be performed using just a local Python script.