lgsvl/simulator #1404: How can I publish constant throttle to SVL with ROS2

nitesh4146, 2021-05-01:
Hi,

I am trying to drive the vehicle at constant speed. I tried publishing `VehicleControlData` msg on `/lanefollowing/steering_cmd` topic as following - 

```python
message = VehicleControlData()
message.header = header
message.target_gear = message.GEAR_DRIVE
message.acceleration_pct = 1.0
```
However, it doesn't seem to work. The steering values are published but not throttle/acceleration. What am I missing here?
FYI- I am working on the lanefollowing image. 

Thanks in advance!

left4taco, 2021-05-02:
You need to publish it at a constant rate, 100Hz maybe? (need clarification from LGSVL simulators developers). 

In addition, constant throttle != constant speed.

nitesh4146, 2021-05-03:
Sure,but if I can understand how to publish throttle values I might as well be able to figure out constant speed. I couldn't find anything in documentation on how to control the throttle with a ROS/ROS2 node. I know how to do this with PythonAPI, but I want to control it through ROS node. Thanks!

left4taco, 2021-05-03:
@nitesh4146 You need to build a ROS2 package first https://github.com/lgsvl/ros2-lgsvl-bridge

It's a bridge between ROS2 and lgsvl simulator. It listens on port 9090 and relay the state/control messages. You'll also need to select the correct Autopiolet profile and use 127.0.0.1:9090 as the API endpoint when you create the simulation in the web page.