lgsvl/simulator #1405: Simulator shuts down in headless mode + Python API  after 30 seconds

left4taco, 2021-05-02:
This is similar but slightly different to https://github.com/lgsvl/simulator/issues/1396 so I'm opening a new issue. 

The `comfig.yml` I'm using is:
```yaml
headless: true
read_only: false
api_hostname: "*"
api_port: 8181
cloud_url: "https://wise.svlsimulator.com"
```

The simulation profile in the website is 52c9044a-090a-426e-a169-db7ab8f70b14(if you can read from the backend). Basically it's in headless mode, with Python API and Autoware.Auto (Apex.AI).

After I clicked `Run simulation`, the window did turn into a 3D map with a car, and I can control the car with keyboard. However, no matter what I do, it alway shuts down after 30 seconds and go back to the original window(dark blue background with  two rose-color buttons) and the status in the website goes back to `idle`. 

Is there a log that I can upload to help debug? Thanks

EricBoiseLGSVL, 2021-05-03:
Yes, please look for the player.log [here](https://docs.unity3d.com/2019.4/Documentation/Manual/LogFiles.html)

marcusvinicius178, 2021-05-05:
Hi I have experienced close/shutdown in simulator, always when I runned the Python Api with AutowareAi with AD Stack for the reason below:

I forgot to run//activate the Ros-Bridge, before start the simulation. You must always activate on runtime manager the TAB maps and MAINLY the TAB SENSING, which launch the websocket bridge. Otherwise the Simulator will be closed with the warning: "Websocket was unexpectedly closed" or similar.

I have also used Autoware.Auto and this shutdown failure occurred for 2 reasons:

1- when the ROS bridge in configure json Tab was not properly defined or some . json sensor configuration was wrong

2- When I have not launched the ros2-web-bridge on terminal or ros2-lgsvl-bridge. I forgot which bridge worked better. However months ago I tested both and they worked. I have never tested the Python Api with Autoware.Auto, but I feel that this error is more related to the bridge setup than the Python Api functionality. 
I cannot evaluate you log file unfortunately as I am just a student user as you 

I hope it works for you.