lgsvl/simulator #1406: How to create and build a package in autoware ai docker?

mengen100, 2021-05-04:
I'm following this tutorial on LGSVL documentation [https://www.svlsimulator.com/docs/tutorials/create-ros2-ad-stack/](url)
However after running `colcon build --symlink-install` and sourcing the file I'm still unable to find my package installed.

Basically, I followed the steps of creating the directory structure, setup.py and package.xml, and then ran colcon build to build the package. After that I sourced both `local_setup.bash` and `local_setup.bash` in _install_ directory. Still not able to find it when using `rospack list-names | grep lane_following` or using `rosrun lane_following drive`.

Anyone knows how to do it correctly? Please help! Thanks in advance!

[image]

marcusvinicius178, 2021-05-05:
Hi, I believe You are running the command in the wrong folder:

you are on: /home/autoware/**shared_dir/autowaredata**/ros2_ws

The shared_dir folder is used just to share files between your host and the docker image. 

I always build my packages on the folder:
home/autoware/**Autoware**/src $ colcon build
Check if the package was built (optional)
home/autoware/Autoware/src $ rospack find (name of package)
home/autoware/Autoware/src $ roslaunch (name of pkg).launch |or| $ rosrun file.py

 Maybe you should create your ros_ws inside the container. And if you have the problem of lost the data always when you exit (as I had previously), just follow this Answer: 

https://answers.ros.org/question/376512/fork-autoware_ai-repository-and-create-docker-image/#377314

(I have customized a way to build a local container for Autoware.AI that stores my data after logout the container, HOWEVER YOU MUST CREATE THIS ROS_WS INSIDE THE /home/autoware/**Autoware folder** or you will lose the modifications...):

I hope you can solve it :)

mengen100, 2021-05-06:
@marcusvinicius178 
Thanks for your response!!!
I followed your suggestion and tried again, but still not able to find the package installed.
[image]

Could you please share your sample package structure as well as the contents? Much appreciate it! (Below is my package structure)
[image]

marcusvinicius178, 2021-05-06:
Hi @mengen100  I am really sorry I have taken a look in your tutorial. And what you have is a new custom docker image tutorial:

[image]

I am using for my project just AutowareAi (with ROS1 pkgs) or AutowareAuto (with ROS2 pkgs) in isolated docker images (separately). From the image above, I realize the creator of this tutorial mixed ROS1 melodic distro with ROS2 dashing distro. From my point of view this is not a good idea, because could generate in conflict of libraries as well as confusing in sourced distro packages...maybe the reason that you do not find the package with rospack find. The command rospack find is just suitable for ROS 1. For ROS2 packages you can locate them with the cmd:

`$ ros2 pkg prefix (name of your package)`

as described here: https://answers.ros.org/question/288501/ros2-equivalent-of-rospackagegetpath/

Be Aware you need to source the bash before compile your pkg with as described here:

[image]

Afterwards with the pkg compiled (finished as you showed me on terminal) you need to source the the install folder (where your pkg was compiled with the cmd colcon build) as you have already done correctly in the first question you described:

[image]

PAY ATTENTION YOU ARE RUNNING A ROS2 BASED PACKAGE!!!! THEREFORE THE COMMAND IS NOT:

`rosrun lane_following....etc`

In fact you should issue:

`ros2 run {your package} {your node}
`
`In your case ros2 run lane_following drive.py
`

Do not forget the extension **.py**  (the programming language the node is written)
Also **ros2 run** has a space between them, while in ROS1 the cmd rosrun has not space. And please source the correct distro, considering your docker image has ROS1-melodic and ROS2-dashing....if you run source /opt/ros/**melodic**/setup.bash your workspace will enable just ROS1 pkgs, and will not locate the ROS2 pkg lane_following you created. Then run source /opt/ros/**dashing/**setup.bash if you are working with ROS2...

I belive this would be enough to fix this issue.

mengen100, 2021-05-07:
@marcusvinicius178 Oops, so sorry I didn't make it clear enough. I'm using autoware docker 1.14.0 for my project, and I'm using that tutorial just for learning how to create the package. I'm aware that the environment and commands are different from the tutorial since I'm using the autoware docker and that's why I was using ROS1 commands such as `rosrun` and `rospack`.
But it's great that you've made me think again, could it because I used a ROS2 method to create a package in a ROS1 environment?

Still wondering how you created your packages in your autoware docker. Thanks for helping and following up!

marcusvinicius178, 2021-05-07:
No problem.... Actually I still did not create a package in my Dockerfile from scratch. What I have done was just modify one of the autowareai.repos packages locally and uploaded to my GitHub account. Afterwards I have modified the file autowareai.repos file to download the modified pkg from my GitHub and not from Autoware GitHub address to the docker Image. I have never tried to create a pkg in Autoware.Ai image.
But I believe it is pretty simple, with the cmd rospack create package ( I do not remember exactly the cmd line) and set the dependencies of this package in package.xml and Catkin make list properly.
I believe also you should create this pkg inside the folder:
home/autoware/Autoware/src 
And then compile it...
Unfortunately I still did not experienced build a package from scratch inside Docker environment.

mengen100, 2021-05-07:
No worries, thanks for your patience and suggestions which helped me figure out I had been using a wrong tutorial. After some searches, finally I found what I was looking for, shouldn't have followed LGSVL tutorial as it is meant for ROS2 environment.

I just managed to create a summary of notes on how to do it in docker, and even tried running it in shared_dir successfully, hope this would help you and others in some way.

### Create and build a package in autoware docker
1. Creating a workspace for catkin
[http://wiki.ros.org/catkin/Tutorials/create_a_workspace](http://wiki.ros.org/catkin/Tutorials/create_a_workspace)
2. Creating a ROS Package
[http://wiki.ros.org/ROS/Tutorials/CreatingPackage](http://wiki.ros.org/ROS/Tutorials/CreatingPackage)
3. Writing a Simple Publisher and Subscriber
[http://wiki.ros.org/rospy_tutorials/Tutorials/WritingPublisherSubscriber](http://wiki.ros.org/rospy_tutorials/Tutorials/WritingPublisherSubscriber)

```
# 1. Creating a workspace for catkin
mkdir -p catkin_ws/src && cd catkin_ws
catkin_make

# 2. Creating a ROS Package
cd catkin_ws/src
catkin_create_pkg <package_name> std_msgs rospy roscpp
# switch back to catkin_ws, source the generated setup file to add the workspace to your ROS environment
cd .. && . devel/setup.bash

# 3. Create a publisher
roscd <package_name>
mkdir scripts && cd scripts
wget https://raw.github.com/ros/ros_tutorials/kinetic-devel/rospy_tutorials/001_talker_listener/talker.py && chmod +x talker.py

# Modify talker.py according your needs.. (it's a vim-like editor I guess) 
rosed <package_name> talker.py

# Building the nodes
cd catkin_ws
catkin_make

# Running the nodes with rescore or runtime manager running in another terminal
rescore
**OR**
roslaunch runtime_manager runtime_manager.launch

# Back to the 1st terminal, run the talker node
rosrun <package_name> talker.py
```

closed 2021-05-10 by mengen100