lgsvl/simulator #1407: Apollo 6.0 and SVL Simulator not working

thymbahutymba, 2021-05-04:
I'm trying to run the simple example described in https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/ without success. I cloned the repository and tried to checkout both the tag v6.0.0 and the branch r6.0.0 (which seems to be a bit more updated). I started the svl simulator and created the new simulation as described in the wiki, compiled apollo, run bootstrap_lgsvl.sh and bridge.sh and so on. Whenever I try to start the localization module, which through dreamview should show me the map, nothing happens. How can I investigate the problem?

Dreamview:
[image]

SVL Simulator 2021.01
[image]

EricBoiseLGSVL, 2021-05-04:
@thymbahutymba Do you have any logs from simulator and can you explain more on what is not working?  It dreamview not working, simulator?

Thinsheep, 2021-05-05:
I had the same issue when I run the default modular testing.
I did modify something like the video "How to run modular testing random traffic simulation from cloud-based web UI"
https://www.svlsimulator.com/docs/tutorials/modular-testing/
then it works I do not know why just try yourself if it works. Otherwise, just load the new apollo.

lemketron, 2021-05-05:
> I'm trying to run the simple example described in https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/ without success. I cloned the repository and tried to checkout both [...]

Use the r6.0.0 branch of Apollo, not the v6.0.0 release tag.

Localization also needs the Transform module to be enabled. Once both are enabled, dreamview should show the map.

thymbahutymba, 2021-05-05:
@EricBoiseLGSVL from where I get the simulator log? Follows the content of the directory modules/map/data.
```
apollohost@apollohost:~/Andrea/apollo$ ls -l modules/map/data/
total 32
drwxrwxr-x 2 apollohost apollohost 4096 mag  4 11:13 borregas_ave
-rw-rw-r-- 1 apollohost apollohost  308 mag  3 14:28 BUILD
drwxrwxr-x 2 apollohost apollohost 4096 mag  3 14:28 demo
-rw-rw-r-- 1 apollohost apollohost 2188 mag  3 14:28 README.md
drwxr-xr-x 2 root       root       4096 mag  3 15:17 san_mateo
drwxr-xr-x 2 root       root       4096 mag  3 15:17 sunnyvale_big_loop
drwxr-xr-x 2 root       root       4096 mag  3 15:17 sunnyvale_loop
drwxr-xr-x 2 root       root       4096 mag  3 15:17 sunnyvale_with_two_offices
```

 @Thinsheep I think that at least the basic example should work as it is. However I will try also your solution. Thanks.

@lemketron ok, I'm using the r6.0.0 branch but still is not working. Actually with using the simulator version 2020.06 it is enough to start the localization module in order to view the map in the dreamview interface. However everything else (once I start all modules mentioned in the wiki the car does not move) is not working since (at least I guess) apollo 6.0 in the 2020.06 is not supported. However I also tried to start both localization and transform and is not working anyway. For "not working" (for now) I mean that I'm not able to see the map from dreamview.

lemketron, 2021-05-05:
> @lemketron ok, I'm using the r6.0.0 branch but still is not working. Actually with using the simulator version 2020.06 it is enough to start the localization module in order to view the map in the [...]

I'm glad to hear that Localization works with 2020.06, though I'm not sure why Localization is not working for you in 2021. What map are you using? Can you try creating a new "random" simulation using the BorregasAve map and Lincoln2017MKZ vehicle, and select the Keyboard sensor configuration for manual control?

With Apollo 6.0, did you select the MKZ LGSVL from both the vehicle and setup menus? If you try the above and enable Transform and Localization, then Dreamview should localize and render the map. With regard to the maps, the Apollo 6.0 (branch) contains only the borregas_ave map. For other maps you'll need to install them yourself into the Apollo tree:
https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/#adding-an-hd-map
 
Getting the car to move can also be a little tricky sometimes... Do you mean it won't move with manual keyboard control or that Apollo won't drive it automatically when you set a destination and enable Control?

thymbahutymba, 2021-05-06:
I just figured out that from version 2020.06 to 2021.01 the "play" button is not pressed automatically so the simulation is not started with just doing "Run simulation" but I have to press "play". So sorry!

closed 2021-05-06 by thymbahutymba

EricBoiseLGSVL, 2021-05-07:
No worries, glad it is resolved and thanks for posting your solution for others.