lgsvl/simulator #1408: [feature request] control an agent in real time

lesun90, 2021-05-04:
I'm implementing an ADV algorithm that focuses on the interaction of the ego with another vehicle/pedestrian. In order to fully test my algorithm, I need to have full control (in real time) of at least 1 vehicle/pedestrian.

For example, I want to test my ADV behavior at a crosswalk scenario,  the ego should if there is a walking pedestrian walking to cross-street. If I control the pedestrian to stop, the ego should continue to drive, if I control the pedestrian walk faster, the ego should stop and wait until the pedestrian crossing... I assumed I can archive the same scenario with the current API, I tried it, but I can get what I need from API, the agent's trigger is not robust for my purpose. 

I am really happy if you already have a plan for this feature, if not, please point me in the right direction so I can work on it myself. I'm assuming we need a ros/cyber agent similar to the ego agent which is controlled via ros/cyber message

EricBoiseLGSVL, 2021-05-04:
We are currently working on being able to make any npc/pedestrian/empty object as an ego.  I think this will solve your issue.

lesun90, 2021-05-04:
Perfect, I am looking for it. Many thanks for developing an awesome tool.