lgsvl/simulator #1415: Some question about "Pedestrian Path Crosswalk" and the position of the buildings

qiongwu1003, 2021-05-06:
Hi,
SVL simulator team

**- Question 1 : "Pedestrian Path Crosswalk"**

I want annotate custome pedestrian  hd map. And I want the pedestrians follow the way as the below picture.
[image]

So I draw the HD map as follow:
[image]

But it dose not work as what I want. No pedestrian go through the crosswalk. Did I do something wrong?
I also download "CubeTown" and "Shalun" maps, but there is no sample like that. So could someone give me some advice.

**- Question 2 : position of the buildings**

How can I get the buildings or other static obstacles position in the code? I have searched the code but I can not find.

Thanks and looking forward to your reply.

EricBoiseLGSVL, 2021-05-07:
You can look at our SF map, we provide an example of a crosswalk that responds to traffic lights.  Are you sure you have the nav mesh set right?

qiongwu1003, 2021-05-08:
@EricBoiseLGSVL Thanks for your reply.

I can not find the SF code in the new source repository. And I checked the old version SF HD map, I could not find the "Pedestrian path crosswalk" sample. Any other way I can check the SF map?

I do the nav mesh follow this [link](https://www.svlsimulator.com/docs/simulation-content/map-annotation/#create-pedestrian-navmesh).
[image]

Also,  any hints on the second question?

EricBoiseLGSVL, 2021-05-11:
@qiongwu1003 Intersection junction 10 is the one to look at
[image]
in the latest version of simulator

Nav mesh is right.  You might have too high of a curb and the pedestrians can't make it.  Can you show a screen shot closer to the curb of the crosswalk at an angle with the nav mesh showing?

Do you have an errors.  What are the pedestrians actually doing.

No we don't have obstacle position but you can make a sensor to get this info.  Position of buildings are based on the origin of the mesh and might not be helpful.

qiongwu1003, 2021-05-11:
@EricBoiseLGSVL 
I am so sorry, I still can not find the SF code in this [link](https://github.com/lgsvl?page=1). Does SF open to custom? If so could you give me a SF link?

Maybe my description is not clear. What I want is for pedestrians to follow the designated route, from A point to B point. 
[image]

I used "CubeTown" to try to do what I want. But I can not make it.
If I do like below(3 lines combined together), the pedestrians will only walk along a single line, and will not cross to the other line.
[image]

If I do like this(1 single line), the pedestrians will walk in the red triangle area.
[image]

**The pedestrians do not follow the HD map lines as the vehicles, do they?**

Also this is the curb with nav mesh:
[image]

Thank you so much.