lgsvl/simulator #1416: OpenDrive export to Apollo map.Apollo map is wrong.

funix0, 2021-05-06:
The Apollo map is rotated 90 degrees counter clockwise.
x = y
y = -x

Why is that

EricBoiseLGSVL, 2021-05-07:
To ensure proper heading values in GNSS related topics, the map must be oriented in such a way the the Z-axis of the unity world coordinate is pointing East. Otherwise all orientation values will have an offset. This can be a bit unintuitive, since by default Unity will have the Z-axis pointing upward on the screen which can lead to users assuming that it aligns to the geographic North.

Please see [here](https://www.svlsimulator.com/docs/getting-started/conventions/)

Is this what you are asking about?

funix0, 2021-05-07:
The exported apollo map is also left-handed ZXY coordinate system. The correct is right-handed XYZ coordinate system. that was i mean.

EricBoiseLGSVL, 2021-05-07:
If you rotate the map correctly and then export do you still have an issue? Unity is left handed, we rotate so it is rotated correctly in apollo and autoware.  I'm confused on the issue