lgsvl/simulator #1417: Depth Camera Output

crixcrixcrix, 2021-05-06:
I am trying to interpret images output from a depth camera. A previous answer says:
> Each pixel in the result records the Euclidean distance between the 3D point corresponding to that pixel and the camera origin
However this does not seem to be the case: values in the depth image are greater for objects closer to the camera, and do not appear to scale linearly with distance.
How do I compute distance from the values in the depth image?

crixcrixcrix, 2021-05-09:
Here are the settings used for the depth camera:
[image]

And here is an example depth image:
[image]

PyTorr, 2021-08-01:
How do you save the Depth image?

zxydi1992, 2021-09-16:
Hi,

I have the same question, from the simulator's code, I guess it is non-linear in  distance.

Sijie-L, 2022-08-18:
Hi,
Any update on this question?  @crixcrixcrix @zxydi1992 
From the [source code](https://github.com/lgsvl/simulator/blob/4c342fd39c5eca6ce6538a144f51d5929812e417/Assets/Shaders/DepthPass.hlsl#L43-L48) I found that the depth is the distance I want but then it goes through some calculations and gets the outColor. Any suggestion on how to decode/transform the int we get from the depth image back to the distance will be really appreciated...