lgsvl/simulator #1418: Possibly wrong linear_velocity conversion in CyberConversions

hoangtungdinh, 2021-05-06:
I'm experiencing that the `/apollo/sensor/gnss/odometry` published the velocity with the wrong orientation. The published velocity seems to be off by +90 degree from the actual velocity. Tracing the code, I found the following lines:

https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberConversions.cs#L475-L480

Should they be:
```
linear_velocity = new apollo.common.Point3D()
{
    x = data.Velocity.z,
    y = -data.Velocity.x,
    z = data.Velocity.y,
},
```

It is because if `data.Velocity` is in the Unity left-handed ZXY, then the conversion to the right-handed coordinate should be defined as explained in this [document](https://www.svlsimulator.com/docs/archive/2020.06/simulator-coordinate-system/#converting-between-coordinate-systems).

EricBoiseLGSVL, 2021-05-07:
Is your map rotated correctly from this issue? #1416

romanodanilo, 2021-05-07:
We are using simulator version 2020.06 and the map assets are the ones tagged as 
`Supported LGSVL Simulator versions: 2020.06`
We are working on BorregasAve, SingleLaneRoad and Shalun assets
btw in dreamview map is also correctly orientated, 
we are experiencing the wrong orientation only in GNSS info. 
Can you help us understand why the `/apollo/sensor/gnss/odometry` has wrong velocity orientation?

EricBoiseLGSVL, 2021-05-07:
This might be a bug.  @hadiTab ?

EricBoiseLGSVL, 2021-05-11:
Thanks for checking 2021.1 release and for finding this issue.  We will look into this.