lgsvl/simulator #1420: How to run NPC vehicles in Python API mode

1Jyfc, 2021-05-08:
Hello!
I use Python API to test Apollo 6.0, and when I create the scenario, I found my ego can run with Apollo, but NPC vehicle cannot run. I want NPC vehicles to run autonomously like random traffic mode, so I need some suggestions.
I refer to [this guide](https://www.svlsimulator.com/docs/python-api/python-api/) and it says:

> Seed (optional) is an Integer (-2,147,483,648 - 2,147,483,647) that determines the "random" behavior of the NPC vehicles and rain effects.

So I use this code:
```
self.sim = lgsvl.Simulator(SIMULATOR_HOST, SIMULATOR_PORT)
self.sim.load(MAP_NAME, 650387)
```
to start `sim`, but it seems useless. 

Here's my code to add a random NPC vehicle:
```
def create_random_vehicle(self):
    sx = self.spawns[0].position.x
    sy = self.spawns[0].position.y
    sz = self.spawns[0].position.z
    angle = random.uniform(0.0, 2 * math.pi)
    dist = random.uniform(20.0, 200.0)
    point = lgsvl.Vector(sx + dist * math.cos(angle), sy, sz + dist * math.sin(angle))
    state = lgsvl.AgentState()
    state.transform = self.sim.map_point_on_lane(point)
    self.sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)
    return
```
and if need, here's my python file: 
[fuzz.zip](https://github.com/lgsvl/simulator/files/6445565/fuzz.zip)

Is there any way to make NPC vehicles in Python API mode to run as those in random traffic mode? Hope for reply, thanks.

EricBoiseLGSVL, 2021-05-11:
simulator/add_random_agents in AddRandomAgents.cs

EricBoiseLGSVL, 2021-05-11:
I see.  No we don't have that functionality.  The api placement logic uses waypoint behavior while the random uses lane follow behavior.  There is no way currently to interchange them.  The Pedestrian and Spawns manager use randomization and needs annotation data to place the pedestrian.  If you want this logic, I would make edits to add agent to pass a parameter to set at a position and then grab closest lane data and add a waypoint to it and start pathing.

1Jyfc, 2021-05-12:
OK I think I need it if it's possible. Thanks.

Actually I think what I need is that, the NPC vehicles can be set at a certain position (which can be recorded by me and I may reuse it after), and then it runs in a legal way by itself. The problem is how NPC runs.
In fact I'm trying to carry on fuzz test of Apollo on SVL, so in a case I need the start and end points of Apollo ego, and start/end point pairs of NPC vehicles.  

For the start/end point pair, now I can only add a random position and fix it onto the nearest lane. Is there any way to add an accurate position point (or get all correct position points on lanes)? Maybe I need Unity to edit the spawns in the map?

And for the running way of NPC vehicles, the random runs by lane follow, but I can't find the logic of how to decide the start point of a random NPC vehicle. I hope if SVL can make NPC run as perfectly as random traffic mode, with a certain start point and several waypoint to decide its path. I tried the waypoint behavior like [13-npc-follow-waypoints.py](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/13-npc-follow-waypoints.py), but it just move simply, and when there're many NPCs, it's more difficult to deal with conflicts than random behavior which acts more realistic and seldom causes conflict.

lemketron, 2021-05-12:
Although there will be a bit of a learning curve, you may wish to check out Scenic which is a higher level (abstract) scenario language that can be used to control LGSVL simulator with Apollo. You can find out more about Scenic here:
https://github.com/BerkeleyLearnVerify/Scenic
https://scenic-lang.readthedocs.io/en/latest/simulators.html#lgsvl

1Jyfc, 2021-05-13:
OK I'll check it. Thanks a lot.

closed 2021-05-13 by 1Jyfc