lgsvl/simulator #1422: how to remove unwanted boundary lines?

Addi2020, 2021-05-10:
How to remove unwanted boundary lines, as shown in the picture below.
[image]

EricBoiseLGSVL, 2021-05-11:
remove from hdmap annotation before running hd map mesh tool

Addi2020, 2021-05-31:
@EricBoiseLGSVL  This solution  do not work. I want to remove one side of the boundary of  lane and then export HD map but it gives an error that (value cant be null).

[image]

EricBoiseLGSVL, 2021-06-01:
From our research, the hdmap formats we support require boundry lines.  Why are you trying to remove them?

Addi2020, 2021-06-02:
@EricBoiseLGSVL As you can see in the first picture , I just want to remove the boundary lines that overlaps with other lane.  Because in real world there is no such lines. 
picture 1= (the boundary lines that are noted by an orange line are unwanted)

EricBoiseLGSVL, 2021-06-02:
- I'm sorry but I am confused.  Are you trying to edit the hdmap for export or are you trying to edit the hdmap mesh generated?
- Why can't you separate the lines out, build the mesh and then add back the lines for export?  You can even make a copy in editor and delete lines as needed for the mesh generation.  You can also export the mesh from the Unity editor and edit the mesh to remove the lines.
- HDmap mesh tool is not meant to be a high fidelity mesh but a simple mesh for users that don't have the resources to create 3d meshes.
- The HDmap tools are not meant to be a stand alone hdmap tool.  It is for simulation of npcs, peds and export to the AD stacks we support.
- Please explain further what you are trying to accomplish so I can help.

Addi2020, 2021-06-02:
1. i am trying to edit the hdmap for export . (base_map.bin file)
2. how to build mesh and separate lines?  can you explain please? 

You can see the yellow boundary line (left boundary lane of MapLane2 (4) ) , overlap on MapLane2 (3). So i just want to remove that yellow boundary line and then export HD map (base_map.bin file). This base_map.bin file will be use to upload in dreamview.

[image]

EricBoiseLGSVL, 2021-06-03:
So you need to edit the annotation.  Here is an example from Borregas Ave annotation.
[image]

You can see that the Yellow line is split off and a white boundary line is added to be the left of the traffic lane.  Is this what you are trying to do?

closed 2021-07-29 by Addi2020