lgsvl/simulator #1424: How to Trigger/ Switch different motion/local planners in Apollo 6.0 ?

marcusvinicius178, 2021-05-12:
Hi Someone here was able to use different local planners available in ApolloAuto version 6.0 or another previous version?

I am aware they have 4 different planners:

[image]

 I am not sure if all of them are obstacle avoiders planners = local planners) or also mission planners/behavior planners. The fact is that there are some rare issues related to it, of users that attempted to change and trigger these planners in Apollo: https://github.com/ApolloAuto/apollo/issues/2429

However I repeated some of these hints and still did not work (at least for Apollo 6.0). My complete issue is described here:
https://github.com/ApolloAuto/apollo/issues/13763

I would really appreciate if someone knows how to make these available planners useful ( How to properly compile them and set the suitable config files on planning module) I am really stuck on that, and I would like that the Ego vehicle avoided the obstacle, which is not happening as you can check below at the end of this video:

https://www.youtube.com/watch?v=9keDL75MS8M

The car stopped behind the NPC Static Agent forever....and any local planner to overtake it was triggered. That is why I which to launch other planners to see different behaviours and hopely some of these 4 available draws a trajectory to avoid this obstacle.

Thanks in advance

EricBoiseLGSVL, 2021-05-18:
@lemketron ?

marcusvinicius178, 2021-05-22:
Hi @EricBoiseLGSVL I have checked from Apollo Documentation, a nice Framework called **Dreamland**, to test motion planners and algorithms with different scenarios:
https://bce.apollo.auto/introduction?locale=en-us

However I **could not complete the profile registration to get access to the "Task Management" tab** so far. They do not accept cell phones outside China. Also I have no idea from the spots it is written in Mandarin, to select the proper region I am. Have you already used this tool, or know how to create a profile there? 

The complete description with the error pictures are depicted below:
https://github.com/ApolloAuto/apollo/issues/13799

EricBoiseLGSVL, 2021-05-24:
@marcusvinicius178 yes, some features of Apollo may not be available outside China, or has not be open sourced.  You can contact the team on the github but there is no guarantee you will get access.  We will keep an eye out and if/when we will try to integrate into SVL Simulator.

marcusvinicius178, 2021-05-28:
Hi @EricBoiseLGSVL my answer was still not answered I believe. My last comment is not the answer of this issue, just another feature consideration. Sorry to make you confuse.

EricBoiseLGSVL, 2021-05-28:
@marcusvinicius178 you are asking about Apollo planners.  We can keep this issue open for a little while to see if anyone can help but since this is a issues page for svl simulator, we will need to mark this out of scope and close soon.  I recommend posting on Apollo's issue page and even link it here, but we can't keep large open ended questions open like this.

marcusvinicius178, 2021-05-28:
Hi Eric, ok in this way I am going to close the issue. I believe it is not just to switch the planners in Apollo. But also the predictor and evaluator, once Apollo generates the trajectory based on obtacle behavior + scenario environment and finally send these inputs to the decision-making...something like that according to their documentation...

[image]

[image]

I will try to change these codes a little bit as they suggest and see if I get some result. I was used with ROS1-navigation Stack where the user could apply different global/local planners, however for automous vehicles, it seems that is not so trivial and not just the "raw" planner will be the responsible to draw the trajectory, but the other modules described here as well....

closed 2021-05-28 by marcusvinicius178

EricBoiseLGSVL, 2021-06-02:
Good luck and let us know what you find. :)

marcusvinicius178, 2021-06-10:
Ok @EricBoiseLGSVL It is easier than I though to change the planners:
It is required just to change the name of the planner on the file **planning_config.pb.txt** from PUBLIC_ROAD to the one you need (my case Lattice). 

The issue was that I could not visualize the planner working **due EXTREME COMPLEXITY of the San Francisco scenario**. Which made the Localization, Transform, Planning modules do not work properly. 

As you can check in videos below on minute 1:20 the **LATTICE planner worked on Single_Lane_Road scenario** (SImple Scenario):

https://www.youtube.com/watch?v=wOC9skKuzUA

On TASKS panel, specifically on center of the screen in minute 3:30 you will see all the modules on GREEN color, which means they are working properly). Look into Module Delay topics:

[image]

**However when I use the LATTICE planner with SAN FRANCISCO Scenario the planner fails,** as you can check:

https://www.youtube.com/watch?v=a6UKYE7xmMY

In console there are messages telling : 
**"RAW IMU message Delay IMU message is 54423 cycles 524.204 seconds behind current time"
"RAW GPS  message Delay IMU message is 52414 cycles 524.342 seconds behind current time"**

The MODULE DELAY panel has bolded Planning module with RED 
[image]

I believe that this happens because the GPU is intensively working in other complex modules (PERCEPTION) and then allocated its power/cores to deal with the complex scenario. As a consequence the LOCALIZATION module is in second plan, prejudicing the TRANSFORM of data (From GPS coordinate to the Vehicle coordinate). 
**Then what I need is a powerful machine or a machine with better clock frequency or both?
Or is this error not related to the computational resources?**

Please @EricBoiseLGSVL @lemketron which is a suitable machine to deal with this DELAT on Input messages (GPS, IMU) to run in SanFrancisco scenario... 
Could **you specify which Desktop/Machine should I rent and where could I rent?**
The Paperspace machines were not able to deal with (**Tested already V100, P6000, P6000x2, RTX5000x2**)
 Just to solve this issue that happens on Complex scenarios.