lgsvl/simulator #1427: Missing Pedestrian NavMesh

duongtran11, 2021-05-13:
When I hit the play button I got an error "missing Pedestrian NavMesh" and no pedestrian appeared.
[image]

I did bake the NavMesh, you can see it in the picture above.
I can't find answer anywhere, it seems like a really simple problem though

EricBoiseLGSVL, 2021-05-18:
This is an issue we have been working on.  Simulator raycasts to 0,0,0 to see if it can find a nav mesh because unity doesn't have an api to just check the bundle.  You can comment out this code until Unity creates this api or we find a better solution.

PedestrianManager.cs line 167
```
if (NavMesh.SamplePosition(pt, out NavMeshHit hit, 1f, NavMesh.AllAreas))
{
    if (!SimulatorManager.Instance.IsAPI && !Loader.Instance.Network.IsClient)
    {
        var pool = SpawnPedPool();
        if (pool != null)
        {
            if (PedestriansActive)
            {
                SetPedOnMap(true);
            }
        }
        else
        {
            Debug.Log("No pedestrian pool, disabled pedestrian manager ");
            gameObject.SetActive(false);
        }
    }
}
else
{
    var sceneName = SceneManager.GetActiveScene().name;
    Debug.LogWarning($"{sceneName} missing NavMesh at {pt} please create navmesh at this point. Pedestrian manager disabled");
    gameObject.SetActive(false);
}

```

duongtran11, 2021-05-19:
The thing is it used to work. I didn't have this error until I change the MapOrigin easting and northing coordinates. but the transform positions of MapOrigin remain 0 0 0
Do you recommend comment out the above block of code or just line 167?
If comment out the above block of code then I couldn't spawn pedestrians right?

EricBoiseLGSVL, 2021-05-19:
Ok let's look at SpawnInfo object in the scene.  Is this object just above the nav mesh?  If not, do this.

EricBoiseLGSVL, 2021-05-19:
Also, what sensors are you using?  If your sensors "see" the peds then they won't spawn unless you change settings in MapOrigin.  There you can set values to ignore this check.  Can you screen shot your map origin object?

duongtran11, 2021-05-20:
[image]
[image]
[image]
I did try it. Doesn't work.
Double checked, the ground has mesh collider
You can see in the picture above I don't have any sensor
[image]
and this is MapOrigin

duongtran11, 2021-05-20:
Alright, I figure it out. 
[image]
The marked parameter in the picture is maximum distance from the spawn position. It was 1 meter back then. Any number bigger than 3 meters should works.
Can't explain why is it so for now, just document it here

EricBoiseLGSVL, 2021-05-20:
That is very odd that it would need >3.  We'll look into this.

duongtran11, 2021-05-21:
FYI, when I set the maximum distance to 1 meter and output position of the NavMeshHit to console:
`if (Loader.Instance.Network.IsClient)
        {
            return;
        }

        if (NavMesh.SamplePosition(pt, out NavMeshHit hit, 3f, NavMesh.AllAreas))
        {
            if (!SimulatorManager.Instance.IsAPI && !Loader.Instance.Network.IsClient)
            {
                Debug.Log("NavMeshHit position: " + hit.position);
                SpawnPedPool();
                if (PedestriansActive)
                    SetPedOnMap(true);
            }
        }
        else
        {
            Debug.Log("NavMeshHit position: " + hit.position);
            var sceneName = SceneManager.GetActiveScene().name;
            Debug.LogError($"{sceneName} is missing Pedestrian NavMesh");
            gameObject.SetActive(false);
        }`

then position turn out:
[image]

and when I set it to 3 meters it turns out:
[image]