lgsvl/simulator #1428: Unable to Launch the  simulator on ubuntu

akshatamk, 2021-05-13:
Hi ,
I'm new to LGSVL. I was trying to setup the simulator environment with executable file on azure NV-series ubuntu machine.

Ubuntu-18.04
NVIDIA-drivers- TESLA M60
https://www.svlsimulator.com/docs/installation-guide/installing-simulator/         have followed these instructions.
But the simulator windows is not opening, Its just flashes away.

Please Help

duongtran11, 2021-05-13:
You gotta enable simulator file to executable before opening it
Go to the folder where you put the simulator file then open terminal, then type in command 
`chmod +x simulator`

tecena, 2021-05-14:
I tried it  still not launching.
Showing this error 
[image]

akshatamk, 2021-05-14:
@duongtran11 Thank you for the reply

> You gotta enable simulator file to executable before opening it
> Go to the folder where you put the simulator file then open terminal, then type in command
> `chmod +x simulator`

 I tried these steps, but still its not working
[image]
showing as aborted, Please help what might be the issue.

sisaha9, 2021-05-15:
You may have to install the drivers correctly: https://www.svlsimulator.com/docs/installation-guide/installing-simulator/

sisaha9, 2021-05-15:
Have you checked the outputted Player.log ? It usually gives more information

akshatamk, 2021-05-17:
> Have you checked the outputted Player.log ? It usually gives more information

I'm not able to find the program.log file. I have downloaded simulator (executable file)  file from official website.
https://www.svlsimulator.com/ for linux.
This is my folder structure
[image]

akshatamk, 2021-05-21:
> Have you checked the outputted Player.log ? It usually gives more information

I'm not getting where to get Player.log file, I searched
In issue #753 Its mentioned the path to player.log file is ( ~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log)
But I do not have Unity installed since I m not building  from source, I have downloaded executable file  running it for linux.
This is my folder structure for the downloaded simulator file
[image]
Please help

sisaha9, 2021-05-21:
Mine is not from source too. It's under ~/.config/unity3d/LGElectronics/SVLSimulator

akshatamk, 2021-05-21:
> Mine is not from source too. It's under ~/.config/unity3d/LGElectronics/SVLSimulator

This is the log file content

Mono path[0] = '/home/simple/Downloads/svlsimulator-linux64-2021.1/simulator_Dat                                                                                        a/Managed'
Mono config path = '/home/simple/Downloads/svlsimulator-linux64-2021.1/simulator                                                                                        _Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'rdp0': 1364x768 (primary device).
Loading player data from /home/simple/Downloads/svlsimulator-linux64-2021.1/simu                                                                                        lator_Data/data.unity3d
Unable to load player prefs
Desktop is 1364 x 768 @ 50 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.4.18f1 (3310a4d4f880)
[Subsystems] Discovering subsystems at path /home/simple/Downloads/svlsimulator-                                                                                        linux64-2021.1/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil)
[Vulkan init] Physical Device 0x1bf84a0 [0]: "Tesla M60" deviceType=2 vendorID=1                                                                                        0de deviceID=13f2
[Vulkan init] Physical Device 0x1bf5270 [1]: "Tesla M60" deviceType=2 vendorID=1                                                                                        0de deviceID=13f2
[Vulkan init] Selected physical device 0x1bf84a0
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=107
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[... 90 lines cut ...]
[Vulkan init] extensions: name=VK_NV_shader_subgroup_partitioned, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
Vulkan API version 1.1.0 (1.2.155 supported by driver)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[Tesla M60] id=[13f2]
Vulkan device type 2
Vulkan driverversion=[0x730800c0] uint=[1929904320]
[... cut ...]
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1364 x 768 @ 50 Hz
InitializeOrResetSwapChain 1024x768 hdr=0 samples=1
Caught fatal signal - signo:11 code:1 errno:0 addr:(nil)
Obtained 2 stack frames.
#0  0x007f305bc17980 in funlockfile
#1             (nil) in (Unknown)

Please help what is wrong
I have installed Both Nvidia and Vulkan drivers.

AKSHATA-MK, 2021-05-28:
Please let me know what is not correct and what has to be done.
I compared the player.log file with other player.log files that were previously posted in the issues here. But I did not find any mistake.

EricBoiseLGSVL, 2021-06-10:
It looks like Tesla M60 in Unity is not supported.  This is a GPU accelerator?

lemketron, 2021-06-10:
Tesla M60 is Maxwell architecture, older than Pascal in our (base/minimum recommendation) GTX 10x0:
https://technical.city/en/video/GeForce-GTX-1080-Ti-vs-Tesla-M60

akshatamk, 2021-06-14:
> Tesla M60 is Maxwell architecture, older than Pascal in our (base/minimum recommendation) GTX 10x0:
> https://technical.city/en/video/GeForce-GTX-1080-Ti-vs-Tesla-M60

Thank you for the reply.
Then which GPUs should I prefer, I'm setting up the simulator in virtual environment in azure. #236 here its mentioned that it works fine with P100,V100 and M60 GPU. Please suggest which GPUs should I prefer?

akshatamk, 2021-06-14:
hi,
I tried to set up the environment with P100 GPU, But still getting the same issue. The simulator just flashes away.
[image]
Please help

lemketron, 2021-06-14:
Can you try running `vkvia` as mentioned here: 
https://github.com/lgsvl/simulator/issues/1473#issuecomment-854903107

akshatamk, 2021-06-15:
> Can you try running `vkvia` as mentioned here:
> [#1473 (comment)](https://github.com/lgsvl/simulator/issues/1473#issuecomment-854903107)

Thank you for the reply.
I tried vkvia command, its showing this
[image]

I checked presence of mesa-vulkan , Its not available in the system.
[image]

I checked the Vulkan installation, It properly installed.
[image]

I'm understanding what is the issue, please help

EricBoiseLGSVL, 2021-06-21:
Do I need to Install Vulkan-SDK? I think so, right @lemketron ?

zouzoui, 2021-07-25:
我也是在Ubuntu18.04中打开LGSVL运行一下子就闪退
请给我帮帮我

Uploading d9ccbceb1ccbb73134c2b520b2244ba0.mp4…

zouzoui, 2021-07-25:
我也是在Ubuntu18.04中运行LGSVL就闪退
各位可以帮帮忙吗？

https://user-images.githubusercontent.com/87936189/126899870-12a6e80c-7160-4a8f-b8b1-4e4cc57aad3a.mp4

EricBoiseLGSVL, 2021-07-26:
@zouzoui this is a very old version of simulator.  I recommend you update to latest version.  You need to post the player.log to see what the error is in the binary

RedBlack888, 2021-08-18:
`nvidia-smi` 
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.91.03    Driver Version: 460.91.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  Off  | 00000000:01:00.0 Off |                  N/A |
| 26%   33C    P8     2W / 260W |    171MiB / 11016MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1640      G   /usr/lib/xorg/Xorg                 45MiB |
|    0   N/A  N/A    901777      G   /usr/lib/xorg/Xorg                 47MiB |
|    0   N/A  N/A    901907      G   /usr/bin/gnome-shell               65MiB |
+-----------------------------------------------------------------------------+

my player.log as follows.

[Vulkan init] extensions: name=VK_EXT_vertex_attribute_divisor, enabled=0
[Vulkan init] extensions: name=VK_EXT_ycbcr_image_arrays, enabled=0
[Vulkan init] extensions: name=VK_NV_clip_space_w_scaling, enabled=0
[Vulkan init] extensions: name=VK_NV_compute_shader_derivatives, enabled=0
[Vulkan init] extensions: name=VK_NV_cooperative_matrix, enabled=0
[Vulkan init] extensions: name=VK_NV_corner_sampled_image, enabled=0
[Vulkan init] extensions: name=VK_NV_coverage_reduction_mode, enabled=0
[Vulkan init] extensions: name=VK_NV_cuda_kernel_launch, enabled=0
[Vulkan init] extensions: name=VK_NV_dedicated_allocation, enabled=0
[Vulkan init] extensions: name=VK_NV_dedicated_allocation_image_aliasing, enabled=0
[Vulkan init] extensions: name=VK_NV_device_diagnostic_checkpoints, enabled=0
[Vulkan init] extensions: name=VK_NV_device_diagnostics_config, enabled=0
[Vulkan init] extensions: name=VK_NV_device_generated_commands, enabled=0
[Vulkan init] extensions: name=VK_NV_fill_rectangle, enabled=0
[Vulkan init] extensions: name=VK_NV_fragment_coverage_to_color, enabled=0
[Vulkan init] extensions: name=VK_NV_fragment_shader_barycentric, enabled=0
[Vulkan init] extensions: name=VK_NV_fragment_shading_rate_enums, enabled=0
[Vulkan init] extensions: name=VK_NV_framebuffer_mixed_samples, enabled=0
[Vulkan init] extensions: name=VK_NV_geometry_shader_passthrough, enabled=0
[Vulkan init] extensions: name=VK_NV_mesh_shader, enabled=0
[Vulkan init] extensions: name=VK_NV_ray_tracing, enabled=0
[Vulkan init] extensions: name=VK_NV_representative_fragment_test, enabled=0
[Vulkan init] extensions: name=VK_NV_sample_mask_override_coverage, enabled=0
[Vulkan init] extensions: name=VK_NV_scissor_exclusive, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_image_footprint, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_sm_builtins, enabled=0
[Vulkan init] extensions: name=VK_NV_shader_subgroup_partitioned, enabled=0
[Vulkan init] extensions: name=VK_NV_shading_rate_image, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_array2, enabled=0
[Vulkan init] extensions: name=VK_NV_viewport_swizzle, enabled=0
[Vulkan init] extensions: name=VK_NVX_binary_import, enabled=0
[Vulkan init] extensions: name=VK_NVX_image_view_handle, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
[Vulkan init] extensions: name=VK_KHR_acceleration_structure, enabled=0
[Vulkan init] extensions: name=VK_KHR_ray_query, enabled=0
[Vulkan init] extensions: name=VK_KHR_ray_tracing_pipeline, enabled=0
Vulkan API version 1.1.0 (1.2.155 supported by driver, 1.1.0 requested)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[GeForce RTX 2080 Ti] id=[1e07]
Vulkan device type 2
Vulkan driverversion=[0x7316c0c0] uint=[1930870976]
Vulkan PSO: cache data not found [path=/tmp/LGElectronics/SVLSimulator/vulkan_pso_cache.bin]
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
[... 8 lines cut ...]

ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
[... cut ...]
FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.
Begin MonoManager ReloadAssembly
- Completed reload, in  0.041 seconds
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
InitializeOrResetSwapChain 1024x768 hdr=0 samples=1
Caught fatal signal - signo:11 code:1 errno:0 addr:(nil)
Obtained 2 stack frames.
#0  0x007f7a90f5c3c0 in funlockfile
#1             (nil) in (Unknown)
                 

@EricBoiseLGSVL  Could you help to sovel this issue? Best

EricBoiseLGSVL, 2021-08-18:
What is this library? `ALSA lib`
Missing audio drivers might be an issue as well.
It seems you are failing to load the gpu.  Have you used Nvidia control panel to set the gpu for svl simulator?

w4c4dwsz, 2021-09-24:
> 我也是在Ubuntu18.04中运行LGSVL就闪退
> 各位可以帮帮忙吗？
> 
>  d9ccbceb1ccbb73134c2b520b2244ba0.mp4

解决了吗老哥