lgsvl/simulator #1429: Simulator on windows and Autoware.AI(ROS) on ubuntu

tecena, 2021-05-13:
Hi,

I want to run the LGSVL with Autoware.AI stack(ROS), currently I have set up the simulator in windows VM. Can I set up the autoware on ubuntu machine and then connect the both. If yes please suggest the procedure to establish the connection between both machines (to run lgsvl with autoware).

Please Help

EricBoiseLGSVL, 2021-05-18:
What have you tried so far and do you have any error logs.  We have a few VM issues that you can search for specific issues.  This issue is too vague.

tecena, 2021-05-21:
> What have you tried so far and do you have any error logs. We have a few VM issues that you can search for specific issues. This issue is too vague.

Thank you for the response.

I have setup the simulator in Windows and the autoware in Ubuntu. Both are running properly in individual machines. Currently simulations are run with keyboard control on localhost:9090 in windows machine. But now to run the simulations with autoware(ubuntu) how to connect the  both. I tried to run the simulations giving the IP address of the ubuntu(autoware) machine along with port numbers. But thats not working.
Please suggest how to connect both simulator and autoware setup on different machines.

EricBoiseLGSVL, 2021-05-21:
Is there any other application using the port you are trying to connect to?

tecena, 2021-05-24:
> Is there any other application using the port you are trying to connect to?

Nope In windows machine where simulator is setup there is no other application running in port 9090. That port is opened for the simulator.

EricBoiseLGSVL, 2021-05-24:
@lemketron any idea why the connection is failing?  @tecena are you starting the bridge before running simulator?

tecena, 2021-05-26:
> @lemketron any idea why the connection is failing? @tecena are you starting the bridge before running simulator?

@EricBoiseLGSVL  Its working fine now . I'm able to connect both thank you

closed 2021-05-26 by EricBoiseLGSVL