lgsvl/simulator #1431: LGSVL quits immediately after starting

komizomakoto, 2021-05-14:
I have same error with this  issue #753.
Please help me anyone.

My environment is as follows:
OS: Ubuntu 18.04
GTX: GeForce RTX 3090
NVIDIA-driver: 460.73.01

I want to use this simulator with AutowareAuto so that I do command ```/opt/lgsvl/simulator``` in
ADE container.
But LGSVL quits immediately after starting.
I have already done command  ```sudo apt remove mesa-vulkan-drivers```.

Player.log is like below.
~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log
output:
```
....
Preloaded 'libvisualization_msgs_interactive_marker_control__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libtest_msgs__python.so'
Display 0 'rdp0': 1920x1080 (primary device).
Loading player data from /opt/lgsvl/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 50 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 0
No supported renderers found, exiting 
(Filename:  Line: 618)
```
And I also show an output of command ```vulkaninfo```. 
```
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Could not get 'vkCreateInstance' via 'vk_icdGetInstanceProcAddr' for ICD libGLX_nvidia.so.0
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
/build/vulkan-tools-KEbD_A/vulkan-tools-1.2.131.1+dfsg1/vulkaninfo/vulkaninfo.h:371: failed with ERROR_INCOMPATIBLE_DRIVER
```
Thank you for your cooperation.

EricBoiseLGSVL, 2021-05-15:
Yes you need to properly install drivers for the GPU.

komizomakoto, 2021-05-17:
Thank you for your reply.

The output of ```vulkaninfo``` I showed was in **ade** container.
The output of ```vulkaninfo``` in host is like below.
```
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.2.70

Instance Extensions:
====================
Instance Extensions	count = 17
	VK_KHR_device_group_creation        : extension revision  1
	VK_KHR_display                      : extension revision 23
	VK_KHR_external_fence_capabilities  : extension revision  1
	VK_KHR_external_memory_capabilities : extension revision  1
	VK_KHR_external_semaphore_capabilities: extension revision  1
	VK_KHR_get_display_properties2      : extension revision  1
	VK_KHR_get_physical_device_properties2: extension revision  2
	VK_KHR_get_surface_capabilities2    : extension revision  1
	VK_KHR_surface                      : extension revision 25
	VK_KHR_surface_protected_capabilities: extension revision  1
	VK_KHR_xcb_surface                  : extension revision  6
	VK_KHR_xlib_surface                 : extension revision  6
	VK_EXT_acquire_xlib_display         : extension revision  1
	VK_EXT_debug_report                 : extension revision  9
	VK_EXT_debug_utils                  : extension revision  2
	VK_EXT_direct_mode_display          : extension revision  1
	VK_EXT_display_surface_counter      : extension revision  1
Layers: count = 8
=======
VK_LAYER_NV_optimus (NVIDIA Optimus layer) Vulkan version 1.2.155, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 0

VK_LAYER_LUNARG_screenshot (LunarG image capture layer) Vulkan version 1.2.170, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_EXT_tooling_info                 : extension revision  1

VK_LAYER_LUNARG_api_dump (LunarG API dump layer) Vulkan version 1.2.170, layer version 2
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_EXT_tooling_info                 : extension revision  1

VK_LAYER_LUNARG_monitor (Execution Monitoring Layer) Vulkan version 1.2.170, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_EXT_tooling_info                 : extension revision  1

VK_LAYER_LUNARG_device_simulation (LunarG device simulation layer) Vulkan version 1.2.170, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_EXT_tooling_info                 : extension revision  1

VK_LAYER_KHRONOS_validation (Khronos Validation Layer) Vulkan version 1.2.170, layer version 1
	Layer Extensions	count = 3
		VK_EXT_debug_report                 : extension revision  9
		VK_EXT_debug_utils                  : extension revision  1
		VK_EXT_validation_features          : extension revision  2
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 3
			VK_EXT_debug_marker                 : extension revision  4
			VK_EXT_validation_cache             : extension revision  1
			VK_EXT_tooling_info                 : extension revision  1

VK_LAYER_KHRONOS_synchronization2 (Khronos Synchronization2 layer) Vulkan version 1.2.170, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_KHR_synchronization2             : extension revision  1

VK_LAYER_LUNARG_gfxreconstruct (GFXReconstruct Capture Layer Version 0.9.6) Vulkan version 1.2.170, layer version 36870
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (GeForce RTX 3090)
		Layer-Device Extensions	count = 1
			VK_EXT_tooling_info                 : extension revision  1

Presentable Surfaces:
=====================
GPU id       : 0 (GeForce RTX 3090)
Surface type : VK_KHR_xcb_surface
Formats:		count = 0
Present Modes:		count = 3
	FIFO_KHR
	FIFO_RELAXED_KHR
	IMMEDIATE_KHR
Segmentation fault (core dumped)
```

Do you think this output also mean that I didn't install drivers for the GPU properly?

EricBoiseLGSVL, 2021-05-17:
Either that or you have mesa drivers installed and they need removed.  Another cause could be that the application is not using the GPU, even though it is installed.  We recommend using Nvidia control panel to force GPU for the Simulator application.

komizomakoto, 2021-05-18:
Thank you for your apply.
1. How can I check that there are not mesa drivers in my PC?
2. Nvidia control panel (Nvidia X server settings) doesn't work now (looks just while screen).
By the way, this command `cat /var/log/Xorg.0.log | grep EE` outputs like below.
Is it cause of driver installed improperly?
```
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 49424.325] (EE) Failed to load module "nvidia" (module does not exist, 0)
[ 49424.325] (EE) Failed to load module "nouveau" (module does not exist, 0)
[ 49424.326] (EE) Failed to load module "fbdev" (module does not exist, 0)
[ 49424.326] (EE) Failed to load module "vesa" (module does not exist, 0)
[ 49424.326] (EE) open /dev/dri/card0: No such file or directory
[ 49424.326] (EE) open /dev/dri/card0: No such file or directory
[ 49424.326] (EE) Screen 0 deleted because of no matching config section.
[ 49424.326] (EE) Device(s) detected, but none match those in the config file.
[ 49424.326] (EE) 
[ 49424.326] (EE) no screens found(EE) 
[ 49424.326] (EE) 
[ 49424.326] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[ 49424.326] (EE) 
[ 49424.327] (EE) Server terminated with error (1). Closing log file.
```

EricBoiseLGSVL, 2021-05-18:
Do you have a monitor connected to this setup?

komizomakoto, 2021-05-19:
Yes I have

lemketron, 2021-05-19:
Was this a clean install of the nvidia drivers? If you had some version prior to 460 then you'll probably want to completely remove them, then install 460 and then reboot.

Does `nvidia-smi` work?

komizomakoto, 2021-05-19:
Thanks for your apply.
My PC is new, I installed Ubuntu 18.04 1 month ago so I think it's clean as for the nvidia dirvers.
But I don't know it's truly clean, so I want to ask how I can confirm whether it's truly clean?

`nvidia-smi` works well like below.
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.73.01    Driver Version: 460.73.01    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 3090    On   | 00000000:01:00.0 Off |                  N/A |
|  0%   36C    P8    11W / 350W |    107MiB / 24265MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2386      G   /usr/lib/xorg/Xorg                 39MiB |
|    0   N/A  N/A      3224      G   /usr/bin/gnome-shell               64MiB |
+-----------------------------------------------------------------------------+
```

komizomakoto, 2021-05-28:
I solved this problem.
The cause was remote desktop app(xrdp)
Thanks for your cooperation.

closed 2021-05-28 by komizomakoto

EricBoiseLGSVL, 2021-05-28:
@komizomakoto can you post any more details on what the issue was and the fix in case any other users have the same issue?

komizomakoto, 2021-05-31:
Thank you for your reply.  
The cause of vulkaninfo error, `Segmentation fault (core dumped)`, was xrdp app which is remote desktop app that is default installed app on Windows OS.  
xrdp did another rendoring from an original rendoring on Ubuntu PC.  
I solved this problem by changing remote desktop app from xrdp to vnc which don't use another rendoring.

EricBoiseLGSVL, 2021-06-01:
Thanks for posting the solution.  I'm sure other users will find this useful.