lgsvl/simulator #1433: Can I import point cloud into LGSVL simulator ?

InterKnight, 2021-05-14:
I have a pcd file which generated by autoware's ndt mapping algorithm, and I wanna import this pcd file into LGSVL simulator as a map. Only pcd file, nothing else. 

I'm wondering if it's workable in LGSVL？If workable, how should I do？If not, what's the file I need to add?

I tryed to import pcd file into Unity and build asset bundle. But when I add the asset bundle as a map in simulator and run it, the car just keep falling down. I guess the reason is absence of ground.

duongtran11, 2021-05-14:
1. Yes you can import point cloud file to lgsvl, either .pcd or .las or .laz work just fine. Go to the Simulator -> Import point cloud. 
2. The car keep falling down because your environment (for this case is your point cloud) need a Mesh Collider component. That mesh collider component define your point cloud as an rigid body object. In the Inspector tab of your point cloud chooose Add Component then find the Mesh Collider
Maybe I mess up some where but you got the idea

EricBoiseLGSVL, 2021-05-15:
There is an option in the importer to create a mesh collider.