lgsvl/simulator #1434: Wrong NPC vehicles moving direction

yejw5, 2021-05-14:
Environments:
```
Ubuntu 18.04
LGSVL 2021.1
Apollo 5.0
```

I have created a BorregasAve test map following steps in https://www.svlsimulator.com/docs/simulation-content/hd-map-mesh-generation/ . 

But in modular testing, the moving direction of NPC vehicles showing on dreamview seems contrarotating with 90 degree.
[image]

Is there some critical steps omitted by me?

EricBoiseLGSVL, 2021-05-15:
Yes, maps are rotated -90 degrees.  See [here](https://www.svlsimulator.com/docs/getting-started/conventions/)

yejw5, 2021-05-15:
> Yes, maps are rotated -90 degrees. See [here](https://www.svlsimulator.com/docs/getting-started/conventions/)

@EricBoiseLGSVL Thanks for your reply. As my understanding, the [Conventions docs](https://www.svlsimulator.com/docs/getting-started/conventions/#map-origin) explains the diff coordinate systems of map view in unity?

But I'm still confused... In my case, with modular testing, it seems that the perception of NPC vehichles and moving prediction rotated (-)90 degrees in dreamview page. And the docs only show a method to switch Unity Editor scene view to a familiar orientation? Is there any detailed steps to fix the moving direction of random traffics in the dreamview page?

yejw5, 2021-05-15:
@EricBoiseLGSVL I have figured it out.... This case can be fixed by rotate the orientation of MapOrigin. Thanks for your reply!

closed 2021-05-15 by yejw5