lgsvl/simulator #1435: Apollo 6.0 car cannot move after resetting active agent

yejw5, 2021-05-15:
Environment:
```
Ubuntu 18.04
Apollo 6.0
LGSVL 2021.1
```

The first time I send routing request, the ego vehicle can run to the destination correctly. But after resetting active agent with `F12` shutcut key, the routing request send from dreamview is not work. To start the simulation, I have to restart Apollo.

However, the problem doesn't exist with Apollo 5.0

EricBoiseLGSVL, 2021-05-18:
Yes, you need to reset Apollo when you reset the vehicle.  Apollo doesn't like a vehicle warping :)

lemketron, 2021-05-18:
It may be sufficient to restart the planning and/or routing modules... one of those is the one that gets upset when the vehicle teleports to a new location.

yejw5, 2021-05-19:
> It should be sufficient to restart the planning and/or routing modules... one of those is the one that gets upset (in Apollo 6.0+) when the vehicle teleports to a new location.

Many thanks. Restarting the planning and routing modules works well!

closed 2021-05-19 by yejw5