lgsvl/simulator #1436: How to get the lidar reflection information?

JinRanYAO, 2021-05-16:
Hello, I run LGSVL with Apollo, but I can't get reflection information from lidar, only can get(x, y, z) , as a result I can't see something such as lanes. How can I solve it?
the lidar information I get(can't see lanes):
[image]
the videos on the Internet by others(can see lanes):
[image]
Look forward to your reply!

EricBoiseLGSVL, 2021-05-18:
What map are you using?  Are you getting intensity from Lidar data? @lemketron @hadiTab

JinRanYAO, 2021-05-19:
Thank you for your reply, I use BorregasAve map. I don't know how to get intensity information from Lidar data, how can I set it up, or where can I get a related tutorial?
Look forward to your reply!

JinRanYAO, 2021-05-19:
> What map are you using? Are you getting intensity from Lidar data? @lemketron @hadiTab

Thank you for your reply, I use BorregasAve map. I don't know how to get intensity information from Lidar data, how can I set it up, or where can I get a related tutorial?
Look forward to your reply!

ntutangyun, 2022-01-24:
hi any update on this ?

EricBoiseLGSVL, 2022-01-24:
No sorry, with Simulator being sunset, no progress will happen with this.  The map needs to have the reflection map texture like borregas in the environmental effects shader we used for the road.