lgsvl/simulator #1437: The NavMesh for pedestrians seems not work

yejw5, 2021-05-17:
I have created NavMesh for pedestrians following [Road network generation -> Creating NavMesh for pedestrians](https://www.svlsimulator.com/docs/simulation-content/hd-map-mesh-generation/#top)

Then, enable`Random pedestrians` and  `Random traffic` in simulation creating. When the test started,  there are non pedestrian walking in the sideroad. Otherwise,  the `Random traffic` option can work correctly.

EricBoiseLGSVL, 2021-05-18:
Do you have a nav mesh at 0,0,0?  Please see this issue #1427

yejw5, 2021-05-19:
I have set position of SpawnInfo to a road mesh, but the random pedestrians still not work. And my building logs have no error message like issue https://github.com/lgsvl/simulator/issues/1427. Is there something I misunderstand?

yejw5, 2021-05-23:
My test use default `Apollo 6.0 (modular testing)` sensor configure. If running on `BorregasAve` map, it works well. So it shouldn't be the wrong sensors.

In the sreenshot of MapOrigin, the nav mesh is set to 0, 0, 0:
[image]

And the SpawnInfo have been above the ground:
[image]

[image]

EricBoiseLGSVL, 2021-05-24:
That spawn point seems too high.  Can you move it closer to the mesh?  We assume you put the spawn info object as close as possible so the vehicle doesn't fall on start.  The check only raycasts for one meter.

EricBoiseLGSVL, 2021-05-24:
Also, do you have any pedestrian annotation?  The pedestrians need annotation to follow in the map.  They don't follow vehicle annotation.

yejw5, 2021-05-25:
> That spawn point seems too high. Can you move it closer to the mesh? We assume you put the spawn info object as close as possible so the vehicle doesn't fall on start. The check only raycasts for one meter.

I have put the spawn info object on the mesh, and still no pedestrian generated. Besides, if setting the spawn point in current height, all functions work well excepting random pedestrian.

> Also, do you have any pedestrian annotation? The pedestrians need annotation to follow in the map. They don't follow vehicle annotation.

Yes, I have created NaveMesh for pedestrians following [docs](https://www.svlsimulator.com/docs/simulation-content/hd-map-mesh-generation/#creating-navmesh-for-pedestrians). Is there any other steps missed?

EricBoiseLGSVL, 2021-05-25:
No the pedestrian annotation, not just nav mesh.  Sorry, this is not clear in the docs, we will update.  Look at CubeTown, it has an example and [here](https://www.svlsimulator.com/docs/simulation-content/map-annotation/#create-pedestrian-path-sidewalk) in docs.

yejw5, 2021-05-25:
> No the pedestrian annotation, not just nav mesh. Sorry, this is not clear in the docs, we will update. Look at CubeTown, it has an example and [here](https://www.svlsimulator.com/docs/simulation-content/map-annotation/#create-pedestrian-path-sidewalk) in docs.

The imported map in my case is downloaded from https://wise.svlsimulator.com/ . Random pedestrians works well in the origin map. And it means that the pedestrian annotation haven't imported completely. So I need to create pedestrian annotation?

EricBoiseLGSVL, 2021-05-25:
Yes, annotation is Unity feature only. You need to create those.

closed 2021-06-24 by EricBoiseLGSVL