lgsvl/simulator #1439: ArgumentException: Getting control 0's position in a group with only 0 controls when doing mouseDrag

duongtran11, 2021-05-17:
This exception or error raises up when I drag the Build window after finishing building simulator.
[image]
Is this something I should concern about? How to fix it or toggle it off?

EricBoiseLGSVL, 2021-05-18:
Yes this is a Unity Editor bug.  Next months release has many improvements for Linux.

duongtran11, 2021-05-19:
Thank you. Is this an autoupdate or how do I update the next month release? I'm using svl simulator 2021.1

EricBoiseLGSVL, 2021-05-19:
We have a patch first and then 2021.2 release.  Probably next month.

duongtran11, 2021-05-20:
You can see that I have already docked the editor window. This error happens only when I drag the Build window (or the Development Setting window) around

EricBoiseLGSVL, 2021-05-20:
Yep, any window (build/dev settings) that is not docked has this issue on linux.  Just a Unity editor bug and you can ignore.  Seems Unity changed some EditorWindow api's for Linux.

duongtran11, 2021-05-21:
I got it, thank you!

closed 2021-05-21 by duongtran11