lgsvl/simulator #1440: Cannot load scene: Invalid scene name (empty string) and invalid build index -1

duongtran11, 2021-05-17:
This error pops out when I hit play button. Below is picture of Scene and Game display. I dont know what is wrong here. Couldn't really find anything helpful in the internet.
[image]
[image]

EricBoiseLGSVL, 2021-05-18:
You are trying to run the scene and this is not supported.  You need to start with LoaderScene.scene and load map from WebUI.  We have DeveloperSettings in the Simulator menu that you can start a sim from the scene but it only loads local assets and is still wip.

duongtran11, 2021-05-19:
I made it works. It run just like in the older version (LGSVL Simulator)
I had to download JaguarXE2015 prefab from the internet then put it to Assets/Externals/Vehicles
Then in the Development Settings that car appears in the ego vehicles
But the Pedestrian NavMesh still missing

EricBoiseLGSVL, 2021-05-19:
You need to place a nav mesh under the spawn info object.  Did you do this?

duongtran11, 2021-05-20:
Here is the solution #1427

closed 2021-05-20 by duongtran11