lgsvl/simulator #1442: Logitech G920 not working with the simulator (Linux)

QZJGeorge, 2021-05-17:
Hello,

I am having trouble connecting the Logitech G920 wheel to the latest release of SVL Simulator. I am using Ubuntu 18.04. The control works fine for the keyboard, but the simulator is not responding to the wheel input. 

I also tried the 2020.06 release of LGSVL Simulator, and the wheel worked perfectly. Therefore, I think it is a simulator built-in issue rather than the system having trouble recognizing the wheel. I have added the wheel control sensor but does not make any difference. 

I have checked this https://github.com/lgsvl/simulator/issues/1329, but that only mentions windows fix. Can someone help me with a linux fix? Thank you very much!

[image]

EricBoiseLGSVL, 2021-05-18:
Yes, we have pending fixes for this sensor for Linux.  We hope to have it released soon. See this issue #1329

Manoj-EK, 2021-08-19:
Hi @EricBoiseLGSVL,

I was curious to know if the pending fixes of sensor for Linux is sorted out!

Thanks in Advance.

Regards,
Manu

EricBoiseLGSVL, 2021-08-19:
I'll check with @lemketron to see if we migrated fix to Linux but I don't we have yet.