lgsvl/simulator #1443: Radar Sensor is not supported in ROS2 bridge

starchrist, 2021-05-18:
Radar sensor simulation is not supported through ROS2 bridge, which is quite a blockage for radar related feature to be simulated in LGSVL.

EricBoiseLGSVL, 2021-05-18:
@hadiTab ?

starchrist, 2021-05-19:
Let me try to describe this issue with more details. I am trying to have a vehicle configured with Radar Sensor plugin from LGSVL library, however, when I am trying to launch that vehicle with ROS2 bridge in LGSVL simulation, the simulation collapse with Error as below:
[image]
It seems that ROS2 bridge is not designed to support radar simulation, is there any plan to have support for Radar Simulation through ROS2 bridge in future?

EricBoiseLGSVL, 2021-05-19:
Yes, this error means that it is not supported.  We added the radar sensor for apollo.  You can take the existing radar code and extend the sensor to include ROS2.

starchrist, 2021-05-24:
Thanks, Eric. Do you have any reference documentation/guideline for having extension of ROS2 bridge to support Radar simulation? Thanks!

EricBoiseLGSVL, 2021-05-24:
No, not any docs on how to support but you can look at our code base and a few existing issues that discuss this.  You need to just create the data conversions so the bridge can serialize the data.

EricBoiseLGSVL, 2021-09-10:
This will be in the next release.  Thanks for posting

closed 2021-09-10 by EricBoiseLGSVL