lgsvl/simulator #1444: Create new scenes and new maps

yufeilongyuan, 2021-05-18:
Hi，Teams：
                I have some difficulties and hope to get some help. I just want to create a new scene, import the point cloud, and then make some roads and lanes according to the point cloud. After I import point clouds, I can't edit roads, etc. But I read our documents, and there are many that I don't need. According to my situation, can you give me some suggestions so that I can make the map I want. Thank you. Hope to recover.

duongtran11, 2021-05-18:
What do you mean you can't edit roads. What roads? And what kind of map you want to make, do you mean 3d environment?

EricBoiseLGSVL, 2021-05-18:
You cannot change point clouds in simulator.  This is not a feature.  If you want a different environment, you will need to create meshes or find a new point cloud that matches what you need.  You can build meshes from imported HDmap annotation also.

yufeilongyuan, 2021-05-19:
OK，I already know what to do, thank you.

closed 2021-05-19 by yufeilongyuan